dresden-elektronik/deconz-rest-plugin #1323: Feature request: integration of Centralite 3-Series Security Keypad

Klagio, 2019-03-04:
From the manual:
"The 3-Series Security Keypad features out-of-the-box compatibility with ZigBee HA 1.2 hubs and devices"

means it can be added easily?

https://www.google.it/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&ved=2ahUKEwj_j7D62OjgAhWF16QKHW1JA34QFjACegQICBAB&url=https%3A%2F%2Fwww.centralite.com%2Fdownloads%2FDataSheet-3SeriesSecurityKeypadMSOs.pdf&usg=AOvVaw1BpDkeQ_uZmGjbjEvGXZWP

Ronoon, 2019-03-07:
First you have to have that one and then you should provide as this link shows:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

It´s impossible to have everything that is on the market for the developers

ebaauw, 2019-03-07:
I don't think this is anything like your regular wireless switch (if there is such a thing), but an IAS ACE device (cluster 0x0501), used to control (arm, disarm, ...) the IAS CIE.  We haven't yet integrated anything like it.  Apart from the technical challenges, I'm not sure what you'd want to expose through the REST API.  We might eavesdrop the commands it sends to the CIE and create buttonevents from those, but it will still control the CIE.  It won't control lights directly through a group.

Klagio, 2019-03-07:
I wish to achieve a command after a successfully typing a specific code: I would use it to disarm alarm or open an external gate. In conjunction with Home Assistant which has deconz component

ebaauw, 2019-03-07:
> after a successfully typing a specific code

As I understand, the keypad does not validate the code - it is passed to the IAS CIE in the _Arm_ command.  It probably makes more sense to expose the CIE instead of the keypad.

> I would use it to disarm alarm 

It should already do that - no need for REST API support.

> open an external gate

How do you control the gate?

Klagio, 2019-03-07:
> How do you control the gate?

the gate is operated with an ESP8266 that is connected to an mqtt server. Home Assistant awaits events from deconz (for example correct code event 1000, incorrect code event 1001), I do a simple automation, when event is 1000 publish a topic  to open the gate, when event is 1001 send a notification with telegram "Attention wrong code entered at your gate".

Pretty easy and powerful

Klagio, 2019-03-07:
From inside the house I do something similar with a Xiaomi switchç press the swith once LOCK  the door, press the switch twice UNLOCK the door.

Of course from the outside I do need a keypad, in order to type a secret code

Klagio, 2019-03-07:
With a Zigbee keypad and deconz/home assistant, there are a lot of cool things that can be done: if centralite allows any entered code could do something (scenes)

Klagio, 2019-03-08:
@ebaauw  are you aware of any other solution to accomplish my goal? I think that would be useful for a lot of people:  keypad of some sort to input a code, that generates deconz_event (succesful code that can be one or more then one, plus unsuccesful code)

ebaauw, 2019-03-08:
> are you aware of any other solution to accomplish my goal? 

Not first hand.  Some of the smart door locks offer integrated keypads, but I wouldn't know if they report unsuccessful code entries.  Typically they report that the door is locked or unlocked.

> keypad of some sort to input a code, that generates deconz_event (succesful code that can be one or more then one, plus an event for unsuccesful code)

From what I understand, you would only get an event with the code that was entered (form the _Arm_ command sent by the keypad).  It's up to you home automation system to determine whether the code was valid or not.

Note that I've made a lot of assumptions on how the Centralite keypad might work, based on the document you linked and on the ZCL spec.  We really need to pair one of these with deCONZ, get the screenshots mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support, and find out what commands the keypad sends, using debug logging and/or Zigbee sniffing.

Although: looking at the Smartthing integration, my assumptions aren't that far off: https://github.com/mitchpond/SmartThingsPublic/blob/master/devicetypes/mitchpond/centralite-keypad.src/centralite-keypad.groovy
Looks like the keypad is reporting temperature as well.

Klagio, 2019-03-08:
> From what I understand, you would only get an event with the code that was entered (form the _Arm_ command sent by the keypad). It's up to you home automation system to determine whether the code was valid or not.

yes understood, in this case wil  have just two events: succesful (armed/disarmed?) and unsuccesful (wrong code).

For opening a door its enough: for scenes of course need other product like this one (for which one scene could be opoening the door of course)

http://www.icasa.io/downloads/

ebaauw, 2019-03-08:
> http://www.icasa.io/downloads/

These are supported, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1124.  The actually send _Recall Scene_ commands to bound lights, so they work even when the gateway is down.  The REST API plugin eavesdrops on these commands and translates them to buttonevents for use in rules (or home automation systems).  If you don't have a Zigbee lock that supports scenes, you'll have to use the buttonevents to open the door.

Klagio, 2019-03-08:
> These are supported, see #1124. The actually send _Recall Scene_ commands to bound lights, so they work even when the gateway is down. The REST API plugin eavesdrops on these commands and translates [...]

yes that could work, although  not perfect (and expensive, 70 Euro!!): I would need to make an automation in my Home Assistant that when (for example) Scene 3, Scene 4, Scene 8, Scene 2 are recalled (sequencially) thats an UNLOCK code.

For opening the door better a binary type (like alarm disarm and code incorrect)

ebaauw, 2019-03-08:
> I would need to make an automation in my Home Assistant that when (for example) Scene 3, Scene 4, Scene 8, Scene 2 are recalled (sequencially) thats an UNLOCK code.

I don't think that will work..  The switch sends broadcast (groupcast) commands, which get repeated by routers throughout the mesh network.  It's recommended to limit the number of broadcasts to one per second.  Using four of these within a second to enter a code will likely cause some presses to get lost due to network contention.

mysteryan, 2019-03-09:
I’ve made a request for a similar class of device (IAS ACE), a key fob a while back.  Unfortunately I’ve only ever gotten two button to report on deconz (Normal/Home and Panic).

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1085
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1139

closed 2019-07-14 by stale[bot]