dresden-elektronik/deconz-rest-plugin #1324: Lights paired but not showing up

jkalousek, 2019-03-04:
I'm trying to run Deconz in docker via [marthoc/docker-deconz](https://github.com/marthoc/docker-deconz).
My problem is that I can see light in windows deconz app but they cannot be discovered using Phoscon app in docker.
Here are steps that I took when trying to resolve the issue:
1. Reset light (3 pcs)
2. Reset Gateway From phoscon App
3. Removed everything from data folder
4. setup gateway from scratch in Phoscon
**Now devices DB looks like this:**
```
id,mac,timestamp,nwk
1,00:21:2e:ff:ff:03:b0:c5,1551714554,0
```
5. Search for lights (no light found according to Phoscon app) but all three lights blinked like if they were paired (and they do not blink anymore after power up - which indicates that they are in fact paired)
**Now devices DB looks like this:**
```
id,mac,timestamp,nwk
1,00:21:2e:ff:ff:03:b0:c5,1551714554,0
2,00:15:8d:00:02:4f:c8:61,1551714712,32596
3,00:15:8d:00:02:4f:c8:10,1551714730,23883
4,00:15:8d:00:02:4f:38:0d,1551714748,35264
```

But still they do not show up in web GUI nor in paired Home assistant
Complete debug log of whole process is attached
[DeconzLog.txt](https://github.com/marthoc/docker-deconz/files/2927124/DeconzLog.txt)

- I'm using these lights: https://www.immaxneo.cz/en/home-2/ (GU10)
- ConBee FW version is 262f0500 (I also tested latest version with same result)
- I alredy raised question here: [Device discovered but not showing up. #107](https://github.com/marthoc/docker-deconz/issues/107) but I was redirected here as it is probably not fault of docker version.
- I did test ConBee with ZHA integrated in Home Assistant (on same machine with same usb cable etc.) where lights are discovered and I can control them so communication works, but there are other SW bugs and I would rather use deconz.

manup, 2019-03-04:
Usually lights should just work, perhaps these here need some code changes.
Can you provide some screenshots of the clusters and the basic cluster after reading the attributes as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

jkalousek, 2019-03-04:
@manup When I use windows Deconz app light are discovered and I can see them in Phoscon too.
Here are some screenshots:
[image]
[image]
[image]
[image]

manup, 2019-03-04:
Indeed hmmm strange, lights also look very valid.

Have you tried creating a backup in the Windows installation and import that on the Docker installation?

The database in the backup contains everything needed for the Phoscon App to show them.

jkalousek, 2019-03-04:
When I transfer database from windows to docker I cannot login to phoscon for some reason...
It just says Login failed.

manup, 2019-03-04:
>When I transfer database from windows to docker I cannot login to phoscon for some reason...
It just says Login failed.

Sorry, yes I forgot about that. You need to reset the password on the login page (currently Windows/Linux have different format for passwords).

jkalousek, 2019-03-04:
Ok, done. I can see group that I created under windows but sadly no lights.

jkalousek, 2019-03-04:
Here is complete debug Log with DB transfered from windows:
[DeconztDebugLog2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2927676/DeconztDebugLog2.txt)

manup, 2019-03-04:
Looks a bit like the Phoscon App might throw a Javascript error while trying to display the lights.
Can you open a developer console in the browser via F12 open the console tab and Refresh the page.

If there is a exception it should be visible in the console with red color.

jkalousek, 2019-03-04:
Thank you for the tip but I already check this in my initial post

> EDIT: Just to add there are no errors in JS browser console.

Only thing that console shows is:
```
global.js?0e6f8d:3 apikey = <REDACTED>
global.js?0e6f8d:3 activated theme Bright
poll.js?240fba:3 add poll repeated lights
websocket.js?e9ddc6:3 websocket connected: ws://192.168.1.112:444/
```

jkalousek, 2019-03-04:
JS is sending request at:
`http://192.168.1.112:88/api/<REDACTED>/lights?_=1551727824478`
and returned data are:
`{}`

manup, 2019-03-04:
Odd this means no light resources are loaded, can you send the backup file to mpi@dresden-elektronik.de I can check if the lights are in there tomorrow.

You may also VNC login into the container, when start light search and read basic cluster, which normally should also recreate the resources.

jkalousek, 2019-03-04:
I will send you backup to mentioned email address but I already check DB and lights are present in both tables "devices" and "nodes".
I will check the VNC functionality.

jkalousek, 2019-03-04:
Well this is what I can see over VNC
[image]

jkalousek, 2019-03-04:
To be complete this is my config of docker image, maybe there is something that I missed:
```
  deconz:
    image: marthoc/deconz
    container_name: deconz
    network_mode: host
    restart: always
    volumes:
      - /docker-data/deconz:/root/.local/share/dresden-elektronik/deCONZ
      - /etc/localtime:/etc/localtime:ro
    devices:
      - "/dev/ttyUSB_ZIGBEE:/dev/ttyUSB0:rwm"
    environment:
      - DECONZ_WEB_PORT=88
      - DECONZ_WS_PORT=444
      - DEBUG_INFO=1
      - DEBUG_APS=1
      - DEBUG_ZCL=1
      - DEBUG_ZDP=1
      - DEBUG_OTAU=0 
      - DECONZ_VNC_MODE=1
      - DECONZ_VNC_PORT=5900
      - DECONZ_VNC_PASSWORD=<REDACTED>
```

manup, 2019-03-06:
Thanks, I've loaded deCONZ with your backup file, turned out the lights fall into a filter which should detect Xiaomi endpoints which happens to look very similar.

jkalousek, 2019-03-06:
Thank you for the information! So It is just a case of adjusting those filters right?

manup, 2019-03-06:
Right, the commit mentioned above fixes the issue https://github.com/dresden-elektronik/deconz-rest-plugin/commit/72006cccc7d79862dbbe9bb857838b4e8c23269c

jkalousek, 2019-03-06:
Thank you for your help. Is there any timeframe for next release were this could be included?

manup, 2019-03-06:
Current plan is this weekend, the release is overdue already but some other work came along.
As alternative you may also compile the latest master branch to get the setup working sooner.

jkalousek, 2019-03-11:
Tested and seems to be working OK. Thank you.

closed 2019-03-11 by jkalousek