dresden-elektronik/deconz-rest-plugin #1325: Missed buttonevents

gazorg, 2019-03-05:
I have a strange (to me) issue with deconz/rest. I have a new philips lightstrip and a dimmer switch. I do get buttonevents from the button, but not after I pair the button with the lightstrip. This is quite annoying as I have a small own program that I also want to react to the buttonpresses. This works for a couple of other combinations of lights and switches. 

Is this a known problem? Is there anything I should try to get this to work?

Philje123, 2019-03-07:
Hello,

I also have this ussue.

I have 2x Ikea wireless dimmers in the bedroom and 2x Osram e27 colour lamps.

As soon as I pair the bulbs to the dimmers in the Phoscon page they no longer send button events via the API and nothing shows in home assistant. The dimmers still show in home assistant and report the battery level and they still show as connected in Phoscon and control the bulbs ok.

gazorg, 2019-03-07:
Hi again. 

I got it to work, sort of. By repairing a number of times it worked or not but eventually I got it to work so that I receive most rest events for buttons and the buttons also have effect on the light. That said, I had to use Phoscon's mechanism to map buttons to effects to get it to control the group that contains the lightstrip (only the lightstrip so far). 

I've had similar behaviour and wonder if there is anything to do that could mitigate this problem.

closed 2019-07-12 by stale[bot]