dresden-elektronik/deconz-rest-plugin #1326: error: 3 resource, /lights/1, not available

ebaauw, 2019-03-05:
The Trådfri colour bulb on my test network went MIA.  It shows red in the deCONZ GUI, and it won't revive when pressing `0` or `L`.  When trying to change it through the REST API, it gives an `error: 3 resource, /lights/1, not available`.  The resource is available when doing a GET though (GET `/lights` as well as GET `/lights/1`).  `state.reachable` shows as `false`.

This is my test network, with only the Trådfri bulb, two XBees, and four wireless switches (one Xiaomi and three iCasa).  The bulb used to be connected to all three routers, with green lines.

The funny thing: the bulb is still reacting to the switches (which are bound to a group of which the bulb is a member), the bulb still sends attribute reports to the gateway, and the REST API plugin sends web socket notifications for the changed light state.  The changes state change is also reflected when GETting the resource.  But somehow deCONZ doesn't reset `state.reachable`.

I haven't yet power-cycled the bulb, in case you want me to sniff something.  I might also try to see if XBees can reach the bulb by unicast, but that will have to wait until the weekend.

manup, 2019-03-05:
Very interesting, a sniffer log would sure be helpful to see what's going on.

I wonder if the gateway tries to reach the light or issues any route requests and how the light responds.

Also interesting is the small network size, the other current Ikea issues let me think that this might only happen in larger networks.

ebaauw, 2019-03-06:
Started the sniffer this morning, but couldn't look at it until now.  By now, the light no longer reacts to the switch.

I stand corrected, the light does not seem to send an attribute report; apparently, the REST API plugin updates the light state and sends the web socket event because it's eavesdropping on the group command.  However, it doesn't seem to updated the group state.  Also, there's no web socket event for the group state.

I tried the ZDP requests from the deCONZ GUI (left drop down).  I see deCONZ sends them and there seems to be a low-level ACK (the one the Zigbee encapsulation protocol doesn't show the details of).  The REST API plugin cannot be tempted into sending anything to the light, and the light is awfully quiet.

Did power cycle the light at 17:30:22.  After that, it responds normally, and the REST API plugin updates the group states.

Here's the sniffer log.  The network key is 0x31356332326132363566393364393964.
[Test Netork.pcapng.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2937229/Test.Netork.pcapng.zip)

manup, 2019-03-06:
The log shows that the IKEA light was completely stuck before power-cycle

    wpan.src16 == 0x15e9

There is not a single command on NWK level before, not even Link Status.
The big question is what killed the light.

----------

Here is a log from my IKEA gateway which gives some insight how they handle their lights:

[IKEA_GW_Reporting3.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2938151/IKEA_GW_Reporting3.zip)

Network key:  f8:f3:2b:e6:70:bf:a9:9b:1e:56:a5:af:16:98:22:72

Since deCONZ 2.05.59 the reporting configuration for On/Off, Level and Color Control clusters is already configured identical as the IKEA gateway, resulting in no periodically reporting.

More differences observed by sniffing the IKEA gateway:
* Configure reporting of Basic cluster ZCL Version attribute 0x0000 (min interval = 0, max interval = 0)
* Configure reporting of Basic cluster Power Source attribute 0x0007 (min interval = 0, max interval = 0)
* Configure reporting of Basic cluster SW Build Id attribute 0x4000 (min interval = 0, max interval = 1800) .. keepalive?
* Bind request Basic cluster to gateway
* Bind request Groups cluster to gateway
* Bind request Scenes cluster to gateway
* Configure reporting of Scene cluster

The periodically report of SW Build Id is interesting, this can be for sure adapted.

I suspect that IKEA routers (and OSRAM, Xiaomi, innr?) don't like the repeated Mgmt_Lqi_req from deCONZ to query the neighbor table. After all this is nothing they are testing since their gateways don't show a network map or use the neighbor table otherwise. I'll add a option to disable the query so it can be testet if it makes any difference.

Ronoon, 2019-03-07:
Maybe that is the issue with my light (INNRGU10) sometimes hanging because the next light is maybe 10cm away and the Mgmt_Lqi_req is the troublemaker there

ebaauw, 2019-03-07:
> Here is a log from my IKEA gateway which gives some insight how they handle their lights:

Not sure I understand the log.  The gateway has NWK 0x0304?  Why not 0x0000?  You're power-cycling a busch-jaeger light/switch (0x3232) and a Hue light (0x515c), but no IKEA light?

> I suspect that IKEA routers (and OSRAM, Xiaomi, innr?) don't like the repeated Mgmt_Lqi_req from deCONZ to query the neighbor table.

What about Hue?  They don't show a network map either.

I still have an innr gateway and an OSRAM Lightify home gateway lying around.  I could setup a small test network with one bulb of the same manufacturer and sniff the traffic, if that's helpful.

manup, 2019-03-07:
>The gateway has NWK 0x0304? Why not 0x0000?

Like the Hue bridge the IKEA gateway acts as ZLL router not a HA gateway.

>You're power-cycling a busch-jaeger light/switch (0x3232) and a Hue light (0x515c), but no IKEA light?

The BJ switch was the easiest one to get the network key by opening the network via button sequence and joining a spare Hue light :)

From what I understand the IKEA gateway treats all lights equally regardless if they support attribute reporting or not.

>What about Hue? They don't show a network map either.

Indeed but I never seen issues with Hue lights going silent, the implementation seems to be very well done.

>I still have an innr gateway and an OSRAM Lightify home gateway lying around. I could setup a small test network with one bulb of the same manufacturer and sniff the traffic, if that's helpful.

Sure, innr would be very interesting. I haven't sniffed the OSRAM gateway traffic in a while, the gateway and account setup drives me crazy every time.

closed 2019-07-12 by stale[bot]