dresden-elektronik/deconz-rest-plugin #1327: "lastupdated" does not use localtime .

MattL0, 2019-03-06:
After a get request via postman to MY_IP:port/api/apikey/sensors

i get this :

```
        "name": "hue bedroom motion",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2019-03-06T08:46:04",
            "lightlevel": 0,
            "lux": 0
        },
        "swversion": "6.1.0.18912",
        "type": "ZHALightLevel",
```

Here is what i get after : MY_IP:port/api/apikey/config
```
{
    "UTC": "2019-03-06T08:30:35",

    },

    "linkbutton": false,
    "localtime": "2019-03-06T03:30:35",

    },
    "timeformat": "24h",
    "timezone": "America/Montreal",

}

    
```

why  "lastupdated" does not use "localtime"  ? And why "UTC" is different then "localtime''?

Does "lastupdated" use "UTC"  for it's value?

heitec11, 2019-03-07:
I am also wundering about the Timeformat.
My different is only one hour, but also not the correct local time.
    "UTC": "2019-03-07T14:04:03",
    "timeformat": "24h",
    "timezone": "Etc/GMT-1",
    "localtime": "2019-03-07T15:04:03",
The Phoscon App is showing the correct time.
The API is showing the UTC time.

ebaauw, 2019-03-07:
The API uses UTC for all time fields.  It's up to the app (i.c. Phoscon) to translate this into local time.

The obvious exceptions are `/config/localtime` and the `localtime` of a schedule.

closed 2019-03-12 by MattL0