dresden-elektronik/deconz-rest-plugin #1328: Multiple Zigbee devices that I cannot get fully recognised by deconz/raspbee

pivot69, 2019-03-07:
I have some Zigbee devices purchased through Wattle (Homecontrol.no) that I want included in my Raspbee deConz network. 

I've managed to get them recognised/paired by the interface, but so far the only node that seems to be fully identified (although not available in the phoscon app, because still not being supported) is the danalock. Every other device from Wattle is not being fully recognised, but do seem to make some sorts of connections to the other nodes...

I have multiple IKEA float panel added, all working fine, plus many IKEA dimmer remotes.

What can I do to look for solutions for this? Any logs I can check, or does anyone know how to properly identify the devices and their zigbee info? Any advice is much appreciated, I would really want to be able to use deConz and Raspbee for my automation.

[image]

***
Some more info about the devices:

1:
0xE89C is the danalock. No question there, I am aware of the issues with those and that I probably need a custom deconz plugin to (possibly) get it to show up in the app... (Still, I really hope it will receive support soon.) 

[image]

2:
0xA7DC is a magnetic window sensor called "Yoga ZB Window Sensor". I got it recognised/paired, but there is no Cluster info for this - although it has been meshing with some of the other devices and does send some kind of alive-signal when activated/triggered.

[image]

3:
0x155D is a zigbee smoke detector I have been trying to add. It is originally a "Flammex k-sd3-ra43100004704" zigbee smoke sensor. Seems to be recognised but no cluster info for this either.

[image]

4:
The final node (the one at the top) with the identity 0x98E7 is a node I am a bit unsure about. But, it could be another danalock, one that I tried adding a few weeks back but never got as "far" as the one I am currently working on. Same manufacturer code at least.

[image]

***
So... where do I go from here? It would be awesome to get these devices working, if not I will need to use the less advanced gateway from Wattle without any API or advanced rule and events capabilities.

After lots and lots of searching regarding the danalock-issue I am certain that I have the latest version of deconz and the firmware, so it should not be an issue with those. I've set up the raspbee by using the raspberry pi image on the dresden-elektronik site, and it is working with all other devices. 

There's probably some logs I can check to look for any errors or other feedback? Something I can post here for some more help? I am optimistic and would much rather receive some expert-advice from you than to continue trying to understand this alone.

Crossing my fingers and hoping for positive response. Thanks! :-)

Ronoon, 2019-03-07:
Can you please provide some more details about the sensors as described in the Wiki:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

ebaauw, 2019-03-07:
Pairing battery-powered devices is a bitch.  Some improvements were implemented lately, make sure you're on the latest deCONZ version, 2.05.59.  

If you still have no luck in populating the cluster list (on the right pull down circle), you could try to select _Read Simple Descriptor(s)_ from the left pull down circle, while briefly pressing the reset button, moving the magnet, or something similar, to ensure the sensor is awake.

If that doesn't help either, power down your routers (lights, yellow nodes), move the device in direct range of the RaspBee/ConBee and try pairing again.  This should force the device to communicate directly to the gateway, instead of over the mesh network.

pivot69, 2019-03-12:
> Pairing battery-powered devices is a bitch. Some improvements were implemented lately, make sure you're on the latest deCONZ version, 2.05.59.

Thanks for your reply! Yep, I can now confirm that I am on 2.05.59. Phoscon app claimed there was no updates, but after a manual upgrade through ssh I got it up to the current version.

> 
> If you still have no luck in populating the cluster list (on the right pull down circle), you could try to select _Read Simple Descriptor(s)_ from the left pull down circle, while briefly pressing the reset button, moving the magnet, or something similar, to ensure the sensor is awake.
> 

Great! Now I managed to get some more information from the window sensor, but not all. It still doesn't understand the "level control switch". Heres some screenshots:

[image]

[image]

Anything else I should share?

I have not been able to get the smoke detector to do the same....

pivot69, 2019-03-12:
Here's the smoke detector:

[image]

[image]

They both claim to be low on battery, according to decent, but that might not be correct. Both batteries came new with the unit

ebaauw, 2019-03-12:
> It still doesn't understand the "level control switch".

That's because endpoint 0x01 uses profile 0xc0c9, which isn't defined in general.xml.  0x0002 is _Device Temperature_, 0x0005 is _Scenes_.  I don't think you're missing any functionality here.

> They both claim to be low on battery

That's because the _Power Descriptor_ reports _Power Level_ as critical.  You might try and read the attributes of the _Power Configuration_ cluster.

> I have not been able to get the smoke detector to do the same....

You need to get it to wake up long enough for deCONZ to read the simple descriptors.

> Anything else I should share?

See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support.

pivot69, 2019-03-18:
> That's because endpoint 0x01 uses profile 0xc0c9, which isn't defined in general.xml. 0x0002 is _Device Temperature_, 0x0005 is _Scenes_. I don't think you're missing any functionality here.

Ok, good to know. But how can I now get this information into the Phoscon app?
I would like to be able to access it and read the temperature information, and of course also read the status info (active/inactive). 

Any way to have it notify about changes in status? 

I have also configured homeseer (with JowiHue) to control the devices, and this seems to be working fine for the Ikea nodes I have. But to get it synced with homeseer I guess it will need to be visible in the Phoscon app too? 

Or is there another way? I did read something about "binding" the clusters, but haven't figured this out. 

> That's because the _Power Descriptor_ reports _Power Level_ as critical. You might try and read the attributes of the _Power Configuration_ cluster.
> 

Yeah, I have tried that. It won't read anything else, it always replies with critical. But even after changing the batteries it still claims this.

> You need to get it to wake up long enough for deCONZ to read the simple descriptors.
> 

Struggling with this. I have managed to wake it, and it is active (blinking its indicator every few seconds) but deconz won't read anything more from it... I will continue trying, but its getting annoying.

> 
> See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support.

I'll fill inn this for the nodes I have been able to read the cluster info.

closed 2019-07-23 by stale[bot]

bohtho, 2019-08-15:
I’m aware of missing functionality/support in Phoscon bit may I ask the process of joining the Danalock with the zigbee network? @pivot69 So far I have tried to search for lights in Phoscon (because of headless RPi) and then putting a pin in the hole on the Danalock and clicking once but I’m not sure if that would work. Maybe I should have opened network instead? Or does it have to be done in Deconz GUI?
I’m planning to try to control it with the REST CLI plug-in afterwards (until there is proper support).