dresden-elektronik/deconz-rest-plugin #1329: [Request device support] Nue wall plate switches

mredz, 2019-03-07:
I'd like to humbly request support for this device please:
[AU Nue wall plate switch 240V](https://www.amazon.com.au/Approved-Standard-Nue-SmartThings-Automation/dp/B07GX943VP/ref=pd_sbs_469_10?_encoding=UTF8&pd_rd_i=B07GX943VP&pd_rd_r=82adb09e-412f-11e9-bdd3-29931c6aa831&pd_rd_w=pXJSj&pd_rd_wg=WK3bS&pf_rd_p=fb083f26-19ee-4869-ae16-5da722448bb4&pf_rd_r=CVY4YJ1TV2DNN8QG9FJA&psc=1&refRID=CVY4YJ1TV2DNN8QG9FJA)
Device support for these would be huge for Australia as these are the only Australian government approved Zigbee wall plates at a reasonable cost. (I would imagine support for these would sell a lot of Dresden's products)
I have managed to connected the device into Phoscon and it comes up as a light. Changes made from the switches are correctly reflected in Phoscon. However, controlling the switches using Phoscon doesn't work. Please let me know how I can assist to get this product compatible?

[image]
[image]

mredz, 2019-03-08:
These devices are discussed on the home assistant support forum. It looks like someone has had success with Phoscon on a similar device. I have tried adding both as a switch and a light...
https://community.home-assistant.io/t/nue-home-bridge-zigbee-3-0-support/53537

Ronoon, 2019-03-08:
You should provide as this link shows:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

mredz, 2019-03-13:
[image]
[image]

mredz, 2019-03-13:
Further to this the switches turn off correctly once after having first detected the device in Phoscon. After that nothing... Actually the on/off commands do work in Deconz reliably!

closed 2019-07-18 by stale[bot]