dresden-elektronik/deconz-rest-plugin #1330: Can not include new device

zoic21, 2019-03-08:
Hello,

I currently test deconz and I have some issue with join process. I successfully include motion, temperature, vibration and double switch from xiaomi. But I can not include : 
- Philips hue motion
- flood sensor xiaomi

I test hundred times, nearby of deconz key (while for others I was a few tens of meters). 

Few days I test philips hue and some other module (switch, temperature) and it's ok, I do deconz key reset and now I have this issue.

I use USB conbee key on intel nuc with vmware on debian 9 (no gui juste rest server).

Thank in advance for your help.

manup, 2019-03-08:
Do you have only sensors/battery powered devices or also routers?

If you only have battery devices, how many?

zoic21, 2019-03-08:
Only sensors device (on battery), for now there is 4 devices on my network.

manup, 2019-03-08:
Hmm ok 4 should not hit any limit.
Which deCONZ and firmware version are you using?

More points to check:
* When joining devices only keep one Tab of the Phoscon App open (including closing it on other devices)
* If the ConBee is connected directly a USB extension cable often helps to prevent interference

zoic21, 2019-03-08:
I'am on deconz 2.05.59 with firmware 260B0500 and I use USB cable extension with only one tab open on my navigator.

Maybe it's du to interference, Conbee key is on same room that phillipsHue hub and wifi... But I can include 4 devices without issue. I start deconz with --dbg-aps=2 but I don't see any trame when I try to pairing module. I attach log.
[deconzServer.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2945694/deconzServer.log)

manup, 2019-03-08:
Oh that's pretty old firmware, recommend to update to 0x262f0500 via:
 *Phoscon App > Menu > Settings > Gateway*

(Make a backup first via Phoscon App)

zoic21, 2019-03-08:
Is there any way to update manually ? Interface say "I'am up to date"

[image]

manup, 2019-03-08:
Yes the steps 2. and 3. from following Wiki can be used to update the firmware (close deCONZ first).

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Note if -d 0 doesn't work you may check if ConBee has another device id via `GCFFlasher_internal -l`

zoic21, 2019-03-08:
Ok link is for rpi, is that ok for conbee or just for raspbee ?

manup, 2019-03-08:
That's ok they both share the same firmware.

zoic21, 2019-03-08:
Ok thank but there is an issue : 
[image]

I 'am root and  lsof on conbee port (/dev/ttyUSB2) show nothing on it

zoic21, 2019-03-08:
Ok I found I have another FTDI key on my system, after shutdown python on this another key it's OK

zoic21, 2019-03-08:
I can upgrade my firmware, I test tonight if I can include my module.

zoic21, 2019-03-08:
With lastest firmware version all seem ok.

Thank for your help. I continue my test.

closed 2019-03-08 by zoic21