dresden-elektronik/deconz-rest-plugin #1332: Network Map scrolling

martynwendon, 2019-03-08:
Somehow I've ended up clicking on the network map on the deCONZ UI and "moved" it .... and now no amount of scrolling around seems to find the map.

Is there a way to "center" the map, or otherwise find it?

I know I can stop and restart deCONZ and it will re-center itself but that seems extreme!

[image]

ebaauw, 2019-03-08:
> Is there a way to "center" the map, or otherwise find it?

Not as far as I know.  I would welcome such a feature.  I've had to resort to the "extreme" workaround of stopping and restarting deCONZ a couple of times, since I couldn't find the nodes anymore., even after zooming out all the way (scrolling down from the mouse).

martynwendon, 2019-03-08:
Oh dear, I feared as much :-(

Please consider it a feature request if possible!

manup, 2019-03-08:
Also internally a long requested feature :)

In the next weeks deCONZ will get a subtle face-lift to improve the currently cryptic toolbar.
A center and zoom to 100% button should be a useful addition.

martynwendon, 2019-03-08:
Thanks, glad it's on the to-do list.  

Even if it were to simply recenter itself if the main window were to be minimised then maximised or otherwise resized, that would be good enough.

At the moment I am trying to painfully do a scroll / grid search to find it ..... I really don't want to restart if I don't have to!

manup, 2019-03-08:
>At the moment I am trying to painfully do a scroll / grid search to find it ..... I really don't want to restart if I don't have to!

Just curious, what's so painful on restarting deCONZ? Should at most take just a few minutes until everything is discovered again.

martynwendon, 2019-03-08:
No problem restarting as such, I'm just new to deCONZ though and reading through the issues list there's quite some reports about things not working well following a restart so was trying to avoid it if possible.

ebaauw, 2019-03-08:
> Just curious, what's so painful on restarting deCONZ? Should at most take just a few minutes until everything is discovered again.

To name a few:
- Quite often, light and groups states are wrong until all lights have been polled.
- CLIPGenricFlag/Status sensors are set to false/0.
- Rules `timestriggered` are reset to 0.

ebaauw, 2019-03-08:
> In the next weeks deCONZ will get a subtle face-lift 

Could you please also have a look at the OTAU plugin, in particular support sorting by selecting the column header, cf. the _Node List_ panel.

manup, 2019-03-08:
Fair points, I guessfor lights/groups these can be improved by storing the current state to the database (already done but seems not to be complete). And restore this state properly.

`timestriggered` could be stored as well on large interval/restart.

>CLIPGenricFlag/Status sensors are set to false/0.

Here we could add logic to restore the last value when it is fresh enough, but maybe this is not desired and more preferable to get a clean state after restart.

>Could you please also have a look at the OTAU plugin, in particular support sorting by selecting the column header, cf. the Node List panel.

Absolutely

lbschenkel, 2019-03-08:
I second the request to allow sorting by clicking on a header in the OTAU table. :-)

JaumeMuste, 2019-03-13:
Is there an option to redraw the links between devices in a different way? For example, drawing it from the center of the box instead of the corner.

closed 2019-07-18 by stale[bot]

kiat-huang, 2021-01-15:
+1 for centering the view and ideally a "fit in the screen" button.  And/or make the default canvas size a lot smaller - it seems to be about 100x too big, hence can easily lose one's work.  I understand someone may have 100s of devices (I'm about a 1/3 of the way there), but most users I'm pretty certain, will have 10-30 which should be easily findable.

kiat-huang, 2021-01-15:
Is this still on the roadmap as mentioned?  Or is not, won't be, hence closed?

Mimiix, 2021-05-28:
Still have this one open :( @donnib 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2997

Mimiix, 2021-05-28:
@manup It's better to post this in #2997

barts2108, 2021-12-20:
I am not sure of other people also noticed, but to me it seems that the zoom-in and zoom-out with the mousewheel works in the opposite way as nearly all other zoom that I know with mousewheel. For example, a webbrowser (Edge, Firefox)m notepad++, ms paint etc etc when you hold down Ctrl then the mousewheel forward is zoom in and mousehweel backward is zoom out.

In deCONZ, mousewheel forward is zoom out and bakward is zoom in

rEs-84, 2022-12-06:
Might of zoomed or moved the nodes out of the view.
Still no way to reset the map/view back to normal? or locate a node on the map from double clicking it from the list?