dresden-elektronik/deconz-rest-plugin #1333: Conbee and multiple channels

PrimaryInc, 2019-03-08:
Hi,

Question?
Because on the size of my Wifi network and use of multiple Wifi channel to not overlap, would that be smart on zigbee as well?

Example:
Wifi Channels 1, 6, 13
Zigbee channels 15, 25 and 26

Will this work or will it be a complete mess?

bdickhaus, 2019-03-10:
I chose to use only two wifi channels, 1 and 6, and I use Zigbee channel 25 for my primary network. I also have a small test network on channel 15. Unless you have a large number of devices that require separate networks, it is easier to manage one Zigbee network using one channel.

PrimaryInc, 2019-03-11:
@bdickhaus, I under stand. 
I do have 4 AP in my and outside my house so 2 channels is to little. I'm in prosess of testing a bit, but it can take a while.
 I have 107 zigbee devices today, and look at adding 40-50 more. And at the time being I'm struggling a bit getting my zigbee network functioning optimally. 

Thanks the bases of my question.

closed 2019-07-16 by stale[bot]