dresden-elektronik/deconz-rest-plugin #1335: DeConz group of lights and Home Assistant integration

WhistleMaster, 2019-03-08:
I have an issue that I don't know how to debug... I have the integration between Home Assistant and Phoscon App and a setup consisting only of Philips Hue stuff: lights and motions. 

Some rooms have a group of 5 lights and others only a single light. I use Home Assistant automation to turn on lights when there is a motion in the room. No issue at all with the rooms with single light. However, with the group of lights, it's kind of hit or miss... Sometime, when there is a motion, the group of lights is turning on but sometime, I can see the motion status "Detected" and the lights status "on", on the Home Assistant GUI, but in fact, the group of lights is not turning on and stays off.

I have the issue only with multiple lights in DeConz group but not with single light in a group.

Thanks for any help !

chimpy, 2019-03-08:
Hey,

I use HA with a deconz group of lights with no problems. Does calling it from the Services page in the HA UI work reliably?

WhistleMaster, 2019-03-09:
No issue at all by calling it from the HA Lovelace UI or the Services page.

chimpy, 2019-03-09:
Have you tried asking for help in the HA forums?

WhistleMaster, 2019-03-09:
Not yet. I had no issue before when I was using the Hue Hub with HA directly. This issue started when I moved to DeConz. That's why I wondering if the issue is with HA or with Deconz API.

Kane610, 2019-03-09:
Unless something wonky is done with the api the component shouldn't be the issue here

WhistleMaster, 2019-06-16:
I re-open this issue as the issue seems not to be on Home Assistant side, thanks to @Kane610' s work. See https://github.com/home-assistant/home-assistant/issues/24122.

WhistleMaster, 2019-06-18:
I'm wondering if the ConBee II could help with this group light issue ? I don't know how to debug that issue, I'm ready to provide any logs if needed.

WhistleMaster, 2019-06-21:
@manup Some Deconz logs when the issue is happening: 

```
18:50:02:871 no button map for: SML001 ep: 0x02 cl: 0x0406 cmd: 0x0A pl[0]: 000
18:50:02:871 ZCL attribute report 0x0017880102011E0F for cluster 0x0406, ep 0x02
18:50:05:672 no button map for: SML001 ep: 0x02 cl: 0x0402 cmd: 0x0A pl[0]: 000
18:50:05:672 ZCL attribute report 0x0017880102010DC7 for cluster 0x0402, ep 0x02
18:50:29:566 ZCL attribute report 0x00124B0018DF3DC7 for cluster 0x0006, ep 0x08
18:50:34:704 no button map for: SML001 ep: 0x02 cl: 0x0406 cmd: 0x0A pl[0]: 000
18:50:34:704 ZCL attribute report 0x0017880102010DC7 for cluster 0x0406, ep 0x02
**18:50:34:785 delayed group sending**
18:50:34:848 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000
18:50:34:848 ZCL attribute report 0x0017880102010DC7 for cluster 0x0400, ep 0x02
18:50:44:659 no button map for: SML001 ep: 0x02 cl: 0x0406 cmd: 0x0A pl[0]: 000
18:50:44:659 ZCL attribute report 0x0017880102010DC7 for cluster 0x0406, ep 0x02
18:50:48:196 no button map for: SML001 ep: 0x02 cl: 0x0406 cmd: 0x0A pl[0]: 000
18:50:48:196 ZCL attribute report 0x0017880102010DC7 for cluster 0x0406, ep 0x02
18:50:49:300 0x001788011027E641 level 255 --> 254
18:50:50:484 0x001788011027DBE3 level 255 --> 254
18:50:51:684 0x001788011027B992 level 255 --> 254
18:50:52:789 0x001788011027E6AB level 255 --> 254
18:50:53:190 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000
18:50:53:190 ZCL attribute report 0x0017880102010DC7 for cluster 0x0400, ep 0x02
18:50:54:053 0x001788011027DBD4 level 255 --> 254
18:50:58:373 Current channel 20
18:50:58:405 Device TTL 4260 s flags: 0x7
18:50:58:763 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
18:50:58:764 ZCL attribute report 0x00158D00025EEA62 for cluster 0x0402, ep 0x01
18:50:58:838 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
18:50:58:838 ZCL attribute report 0x00158D00025EEA62 for cluster 0x0405, ep 0x01
18:50:58:934 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
18:50:58:934 ZCL attribute report 0x00158D00025EEA62 for cluster 0x0403, ep 0x01
```

Could it be related somehow to **18:50:34:785 delayed group sending** ?

WhistleMaster, 2019-06-27:
Still having the issue... Any help to debug it would be really appreciated ! Thanks !

manup, 2019-06-27:
Can you please provide the details how the rules (?) in your Home Assistant setup are done.
Basically which commands are transferred to the REST-API when the motion sensor is triggered.

WhistleMaster, 2019-06-30:
Thanks for your reply! I've tried to gather some log when I opened the issue on the Home Assistant side: [home-assistant.log](https://github.com/home-assistant/home-assistant/files/3268338/home-assistant.log)

I'm not sure where I can catch which commands precisely are transferred to the REST-API, maybe @Kane610 could help. 

I can share the YAML rules though:

```
- alias: Bureau Lights on based on motion
  initial_state: 'on'
  trigger:
    - platform: state
      entity_id: binary_sensor.motion_bureau
      to: 'on'
  action:
    - service: light.turn_on
      entity_id: light.bureau
```

In HA:
[image]
[image]

In Deconz:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Kane610, 2019-06-30:
You should be able to get most things from debug prints

logger:
  default: info
  logs:
    pydeconz: debug
    homeassistant.components.deconz: debug

WhistleMaster, 2019-06-30:
Thanks for your reply. 

I already had the following:

```
default: error
logs:
  pydeconz: debug
  homeassistant.components.deconz: debug
```

My log were taken with that configuration.

Kane610, 2019-06-30:
You are referring to the logs in the other issue right?

WhistleMaster, 2019-06-30:
Yes indeed. The log file in my post just above is taken from it.

WhistleMaster, 2019-07-09:
Do you need anything else to debug the issue ? Thanks !

WhistleMaster, 2019-07-21:
I guess this issue will stay unfixed... It's really a shame to not be able to use groups of lights...

WhistleMaster, 2019-07-21:
@manup Do you need anything else to debug the issue ? Thanks !

WhistleMaster, 2019-07-21:
I still have the issue, but I've made some tests, and I've tried to use Phoscon only to do the same (detect motion and light the group of lights) and it works perfectly, no issue at all...  

I suspect then that the issue is with the integration between Home Assistant and Deconz.

Kane610, 2019-07-21:
You said calling the hass service works perfectly fine controlling the groups. Can you share what you send that works?

WhistleMaster, 2019-07-22:
> You said calling the hass service works perfectly fine controlling the groups. Can you share what you send that works?

[image]

WhistleMaster, 2019-07-22:
Let's me recap there:
- Using Phoscon rules to turn on Deconz group of lights with motion: OK.
- Using HA rules to turn on Deconz group of lights with motion: NOK.
- Using HA rules to turn on HA group of lights with motion: OK.

Kane610, 2019-07-22:
So the toggle service works properly? In the backend it just uses turn_on/turn_off service so it is weird if that one works but not the other

WhistleMaster, 2019-07-22:
Indeed... I really don't understand why but only the following automation, using Deconz group, brings erratic results:

```
- alias: Bureau Lights on based on motion
  initial_state: 'on'
  trigger:
    - platform: state
      entity_id: binary_sensor.motion_bureau
      to: 'on'
  action:
    - service: light.turn_on
      entity_id: light.bureau
```

The issue is that it's not consistent, sometimes it works, sometimes not.

Kane610, 2019-07-22:
I still don't see how the integration can be at fault here, there are no error messages

WhistleMaster, 2019-07-22:
I know but I really don't know what to do to help debugging the issue. Do you have by any chance a group of 5 lights that you can create within Phoscon to test with HA ?

Kane610, 2019-07-22:
Yes, I have one group with 9 lights and a power plug and it works without hiccup. Clicked it multiple times and worked as expected

WhistleMaster, 2019-07-22:
Well, I'm using a motion detector, but that should not be so different as I can see that motions are detected correctly. Could it be the lights brand ? I'm using Philips Hue GU10 in those groups. I have 15 lights like this one, grouped by 5, and the issue is always the same with those groups of Hue GU10.

Kane610, 2019-07-22:
These where all Philips gu10 with tunable white

Kane610, 2019-07-22:
All of mine are on the latest fw as well

WhistleMaster, 2019-07-22:
I have the color one with fw 4.32 build 26420, according to Deconz. How can I make sure that's the latest fw and can it be updated with Deconz ?

Kane610, 2019-07-22:
It can't be updated with deconz, need to do it from their Hue bridge. You can find all device firmwares from Hue here https://www2.meethue.com/sv-se/support/release-notes

WhistleMaster, 2019-07-22:
Well, I don't understand what could be wrong and to debug it... Do you have a ConBee I or II ?

> It can't be updated with deconz, need to do it from their Hue bridge. You can find all device firmwares from Hue here https://www2.meethue.com/sv-se/support/release-notes

So you have to unlink all lights from Deconz and link it back to Hue bridge to be able to update them ?

Kane610, 2019-07-22:
Migrated to a conbee 2 a month ago, upgraded the fw of all devices at the same time

WhistleMaster, 2019-07-22:
Well, this is my last resort, I'll buy a ConBee 2, link back all devices to the Hue Bridge, and link to the ConBee 2. Hope it will fix my issues... Thanks !

Kane610, 2019-07-22:
Good luck!

WhistleMaster, 2019-07-23:
I've made some other tests, by putting Phoscon next to HA to see what is going one, if the status reported by HA and/or Phoscon was correct, and both are not !

After a motion is detected, the lights are reported both by HA AND Phoscon as ON, which is not the case as the lights stay physically OFF. So there is indeed an issue with Deconz/Phoscon and not with the integration with HA.

@manup Therefore, the issue seems to be with the Philips Hue Spot GU10 (version 5.127.1.26420) and Deconz.

[image]

antonbo, 2019-08-24:
I have similar behavior with but with IKEA G10 W lights (8 pieces).
So my observations are:
- Controlling individual lights from either HA or Phoscon Web app work quite reliably
- Controlling groups from either HA or Phoscon Web app work does not work well, especially if I want to turn the off.

HW: ConBee 2 with FW 0x264a0700
Deconz is running in the Docker
HA version 0.97.2
Quite often I could see my lights displayed in the webapp are not synced. They might be show as On while light level is 0 (not sure if they are actually turned off or brightness set to 0).

I could collect the logs for both Deconz and HA. Please share you log settings and what you want me to do.

closed 2019-12-30 by stale[bot]