dresden-elektronik/deconz-rest-plugin #1336: Conbee not working anymore

ni-ka, 2019-03-08:
Hi,

My ConBee USB stick has stopped working. I've tried resetting with GCFFlasher, restart, restart of Raspberri pi, setup on Windows, firmware down and upgrade, but unfortunately I can't detect any lights anymore.

It shows up in my phoscon webapp as connected (262F0500) but detection doesn't find devices. I use Osram Flex Led Strips and they worked fine before.

Here pics from the Network settings on Windows (sorry phone pictures was easier...)

[image]
[image]

Any ideas? Unfortunately I live in Senegal and getting a replacement is a bit of a challenge...

Cheers

manup, 2019-03-09:
Please check the new Zigbee configuration page added in 2.05.59.
Here you can see previous configurations and revert back to them. 

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Check for changes of Channel, PANID and Network key, if these changed at some point of time, load to the previous configuration.

ni-ka, 2019-03-09:
@manup That's very helpful, unfortunately in the process I reinstalled deconz (am using docker) and thus I don't think the history is there. I only see 1 channel change.

manup, 2019-03-10:
Hmm I'm afraid in this case you need to reset the devices and rejoin them. They are currently connected to the former network using the respective channel, panid and network key; without the config it is not possible to recover the network key.

closed 2019-07-15 by stale[bot]