dresden-elektronik/deconz-rest-plugin #1337: Lights getting unreachable too often, but group control works

abmantis, 2019-03-08:
Using 2.05.59, lights are getting unavailable too often. It seems to happen more with lights that have better connection trough other lights (routing) then directly to the RaspBee, so it could be a routing issue. 
However, using the group control (even if it only contains a single light) usually works immediately, even when the light appears as unreachable.

olemr, 2019-03-09:
Ikea lights?

abmantis, 2019-03-09:
@olemr no. actually, they are power plugs from a local company.

olemr, 2019-03-09:
OK, so not just Ikea then. I have had this problem since .29

abmantis, 2019-03-09:
I have an IKEA one, but it is connected directly to the Raspbee, so no routing and no issues.

tamet83, 2019-03-18:
I have the same problem with a philips hue light strip. It’s very annoying because I relay on this device for some automations.

abmantis, 2019-03-26:
@olemr actually, yesterday the Ikea bulb was also affected by this issue.

timbru31, 2019-07-29:
Bump to avoid closure by the stale bot (sorry for e-mail spam), but this is still an issue with the .66 version. Only IKEA lights are affected for me.

zilluss, 2019-11-30:
Bump, very annoying bug, the lights are basically unusable via Phoscon. This occurred suddenly a week ago and only for one group. The state inside the application changes, but the lights inside the group won't turn on or off unless the light group is toggled.

xrmthomas, 2019-12-05:
I see this behaviour as well, but with philips HUE bulbs. I have 2 groups that each have 4 bulbs, and one  (the same) bulb just stops responding every now and then, but group control from the phoscon app does still work. Annoying, as my openHAB setup is totally depends on this.

When using the api through the openhab plugin turning the lights on does work, but it won't turn them back off.

I've done some reboots, reloaded some older configs, tried some other stuff.. The problem sometimes goes away for a while and then suddenly comes back.. I was first thinking of a range-problem, but as these are bulbs that are in the middle of everything that seems unlikely. Also, deconz shows these items as being connected.

closed 2020-04-11 by stale[bot]