dresden-elektronik/deconz-rest-plugin #1339: IKEA Trådfri CWS bulb, firmware updated but old firmware is still listed as current in deCONZ?

mwhrtin, 2019-03-09:
Running deCONZ 2.05.59 and a ConBee stick with firmware 26300500, normally `deconz` is started in headless mode on a Raspberry Pi.

**Issue:**
I suspect all my IKEA Trådfri bulb E27 CWS opal 600lm have been updated to the latest firmware version `1.3.009` that was released about a week ago, the bulbs appear to have new functionality/behavior that they didn't have before. But deCONZ still list them as using the older `1.3.002` version. I've tried to reboot, triggering a manual OTAU doesn't do anything, the progress just changes to `Idle`.

Is it possible that the bulb firmware is updated but that deCONZ (for whatever reason) is still listing the older firmware as the bulbs current version?

**Node list screenshot:**
[image]

**OTAU plugin screenshot:**
[image]

**Longer description:**
IKEA released new firmwares about a week ago, I think I remember reading that deCONZ will eventually do an OTAU in the background if there are new firmwares available, so I used `ikea-ota-download.py` to download them.  I verified that the new firmwares were downloaded and a new `117C-2801-13009572.zigbee` file was generated in the `otau` directory. 

Speed forward to today and I noticed that all 5 of my IKEA Trådfri bulb E27 CWS opal 600lm retain both the brightness and the previously used color when power-cycled. This was not the case before, previously they kept the brightness but the color was always reset to 2700K white when cycling the power using an old fashion power switch.

I suspected that perhaps the bulbs had been updated in the background and this was one of the fixes in the new `1.3.009` firmware for the bulb. After a quick search I found other Trådfri CWS owners with the same experience (color is kept) after updating the firmware. So I assumed that this confirmed that my bulbs were now using version `1.3.009`.

But to my surprise when I checked in `deconz-gui` (and in Phoscon), the lights are still listed as using version `1.3.002`. I've tried to reboot in case the firmware version is only read at startup but `1.3.002` is still listed. When I try to perform a manual OTA update using `117C-2801-13009572.zigbee` the progress is immediately changed to `Idle`, no progress is started. I'm not sure if this is because the update fails or if the bulb rejects the update because it's already updated.

ebaauw, 2019-03-09:
Mine reports 1.3.009 after the upgrade.  Did you read the _SW Build ID_ in the _Basic_ cluster after upgrading the firmware?  Afaik the REST API plugin doesn't poll this attribute spontaneously.  The value in the _Node List_ panel is updated by the REST API plugin.

mwhrtin, 2019-03-10:
Okay, I didn't know they weren't always in sync. In _Basic_ cluster and _SW Build ID_ it reports `1.3.009` so it looks like the bulbs are in fact updated. 

When does the REST API plugin poll this attribute to get a potential updated value? If I understand it correct it's not done during startup/reboot, not after OTAU and not spontaneously. Does it only read it once when the device is registered and never again or can I trigger it to update manually?

manup, 2019-03-10:
I'll add some code so that the *SW Build Id* gets reported every 30 minutes.
In fact the IKEA gateway configures the attribute like this as well, perhaps to have a health check for the light.

KennethLavrsen, 2019-06-22:
I am just reading this bug report and I am shocked.

I have been trying to upgrade Tradfri devices with OTAU. I have waited 4 hours each time. I have checked the revision programmed and assumed it all failed because the version did not update in deconz or Phoscon. Even after the devices have been in operation for more than a month since I GAVE UP the versions are still the old. And now I read this and find out that by some mysterious not very user friendly user interface I can actually read out the software version and the devices were actually OTAU updated. 

How can you get the idea that it is OK to OTAU update and not update the software revision in the database????? And why is this bug still open? It does not update the software revision at all

manup, 2019-06-22:
>I am just reading this bug report and I am shocked.

Please note that the IKEA firmware update is an unofficial feature and not officially promoted.
It requires running a script to download the related firmware files as a first step.

See: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ikea-ota-download.py

    python2 ikea-ota-download.py

* Lights can now be automatically updated, and should usually start the update after a power-cycle
* The remote and sensor can only be updated manually within the deCONZ GUI, this is a choice to not drain battery without user permission

We have a way user friendlier OTA update interface within the Phoscon App in the works but it will take some time until this is available.

>I have been trying to upgrade Tradfri devices with OTAU. I have waited 4 hours each time. 

Which IKEA devices are you trying to update? 

The IKEA lights and plug do send their current software version aka "SW Build Id" on their own, no need to query it manually.

KennethLavrsen, 2019-06-22:
The problem that I highlight is the exact original problem reported (otherwise I would have opened a new bug report).

The software version of the devices I OTAU are not reported. The old version is the one that is reported - everywhere both in Deconz and Phoscon. Even in Deconz right after the OTAU is complete the software version reported is the old. Even after an entire month.

I do not know if it is all devices. But it is surely the case for the IKEA small 2 button on-off switches. I have 4. Two were on 2.0.19. Two on 2.0.020. When I updated the two 2.0.019 they completed the OTAU - and this takes 4 hours for each! And when done and you return to Deconz the UI shows... nothing. It is like if you never did any OTAU. Not even a Done or Complete. And the software reports the old revision from the device. So it failed. You think. So you try again. And again. 4 hours. New battery and again.

So now I need to upgrade all 4 to 2.2.008. New firmware. New Deconz. Hope it works this time. And same problem.  This time I set an alarm clock and got up at 5 AM and stared at the OTAU completing to see what happens. The UI ends with "Done". Wow. And then seconds later Done is gone and you cannot see what happened. So you have to sit and stare at the screen to catch it. Bad!. So I start searching for clues via Google and end up on this bug report. And yes. The device are updated. You just cannot see it.

The horrible user interface is what it takes of "secret" tricks to get it updated. 
You have to know that you must select the device box in the Network view clicking on the small button on the right. You must select the 0000 Basic line (not the OTAU line). You must look at the cluster info tab on the left screen and find a line that says Software Build ID. You must click on this line at the left most column. A pop up comes and you can then click Read. This probably fails many times but works if you ull out the battery of the unit and try a few seconds after.
You have to know about all this steps to get Deconz to read the software revision from the device. I only found out because I found this bug report. 

There are TWO bugs that urgently needs fixing. Both fit the original report.
- First when you OTAU update - Deconz need to write the new software version to the database when the OTAU is completed successfully. It should know right? It just did it!
- Second at least some battery devices like the IKEA two button are not updated even after many weeks of operation. Probably never.

Those two are the urgent things to fix. Otherwise people like me will continue to assume that OTAU is not working.

Second - do something about that UI. 
When OTAU completes after hours of upgrading make sure the user can see it is complete without having to stare at the screen.
And get a more user friendly way to force a query of the device for its information such as software revision. When you look at the list of nodes list on the right side of the screen the software revision and information in the other few columns should be possible to refresh without having to go through all the steps I describe above.

KennethLavrsen, 2019-10-20:
Ping. Go away stale bot. This one needs fixing

KennethLavrsen, 2020-02-17:
Hello stale bot. I am still here

Mimiix, 2020-06-05:
@KennethLavrsen Is this still a issue?

KennethLavrsen, 2020-06-05:
I will check this weekend. It has been a while since I upgraded any devices so I hope I have an IKEA device that is not on latest firmware

Mimiix, 2020-06-05:
Alright, let me know :)

slamp, 2020-06-15:
I confirmed that the old firmware is still listed. I updated the firmware of conbee ii and the old version is deplayed, even after a restart
[image]