dresden-elektronik/deconz-rest-plugin #1342: [Help needed] Cannot get ConBee working

imbaimbaimba, 2019-03-10:
[Hi,
Recently received my ConBee stick, and unfortunately haven't been able to get anything working so far.

What I did was plug in the ConBee stick to my Synology DS1019+, and tried to figure out what it was detected as. I tried all sorts of commands I could find for this.
lsusb gives me:
```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0403:6015:1000 00  2.00   12MBit/s 90mA 1IF  (FTDI FT230X Basic UART DM0103QZ)
  |__1-2         0658:0200:0000 02  2.00   12MBit/s 100mA 2IFs ( ffffffd1ffffffb2ffffffdbffffffad)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00887062CA6030)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
```
and eventually found out that after inserting it and running dmesg I see this:
`usb 1-1: new full-speed USB device number 9 using xhci_hcd`

From there I tried running the deconz docker image from marthoc by:
sudo docker run -d \
    --name=deconz \
    --net=host \
    --restart=always \
    -v /volume1/docker/deconz:/root/.local/share/dresden-elektronik/deCONZ \
    -v /dev/bus/usb:/dev/bus/usb \
    -v /etc/localtime:/etc/localtime:ro \
    -e DECONZ_WEB_PORT=8080 \
    -e DECONZ_WS_PORT=8443 \
    --device /dev/bus/usb/001 \
    marthoc/deconz

This is what I see in the web interface under gateway:
https://i.imgur.com/7E6fiba.png

As far as I understand, it is supposed to show this: ?
http://klein-carsten.de/openhab2/logs/Deconz_Conbee_ready.JPG

I tried with a couple of Xiaomi sensors to open a search and press the connect button on the sensor, but I can't find anything.

I am not sure if I have gotten the device right. I also have an Aeotec Z-Wave stick in the Synology that is working, and unfortunately I do not remember what device I had to pass through my home assistant back when I installed that. I tried to find out the docker run command I had run back then, but I cannot seem to be able to find that out. Though it must exist somewhere as I have Ouroboros (watchtower alternative) running to auto update my dockers, and it must be issuing the same command after every update, as the Z-wave stick is still working.

Anyone that can help me get this going?

Elvys8, 2019-03-12:
Hi!

I have a Synology DS218+ and I'm having problems getting my ConBee working, too. Following your comments I found out that my stick is also plugged to de USB1. I added the value "/dev/ttyUSB1" to the "DECONZ_DEVICE" variable on the Environment tab and now it shows the right icon, but I still get the "Firmware Not connected" message plus a "No connection" error.
[https://i.imgur.com/lCs0QUY.png](https://i.imgur.com/lCs0QUY.png)

Any ideas? Thanks!!

duffbeer2000, 2019-03-30:
Try to run the Container priviledged, then it should work:
--privileged

closed 2019-08-04 by stale[bot]