dresden-elektronik/deconz-rest-plugin #1344: Running deconz on a loxberry

elmocito, 2019-03-10:
I try to run deconz on a [loxberry](https://www.loxwiki.eu/display/LOXBERRY/LoxBerry) and get the following error codes after running _sudo service deconz-gui status_

`Mar 10 22:47:39 loxberry systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Mar 10 22:47:39 loxberry systemd[1]: deconz-gui.service: Service hold-off time over, scheduling restart.
Mar 10 22:47:39 loxberry systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Mar 10 22:47:39 loxberry systemd[1]: deconz-gui.service: Start request repeated too quickly.
Mar 10 22:47:39 loxberry systemd[1]: Failed to start deCONZ: ZigBee gateway -- GUI/REST API.
Mar 10 22:47:39 loxberry systemd[1]: deconz-gui.service: Unit entered failed state.
Mar 10 22:47:39 loxberry systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
`

What is the problem? I read something about a "Pi user", but I cannot add another user to the Loxberry instance.

wlatic, 2019-03-14:
Do you have the log where you saw the pi user error?

Also please provide the start up script you are using and the username you want to run deconz under.

MBeeken73, 2019-03-16:
Hat jemand Deconz lauffähig auf einen loxberry?

elmocito, 2019-03-16:
> Do you have the log where you saw the pi user error?
> 
> Also please provide the start up script you are using and the username you want to run deconz under.

Where do I get that?

closed 2019-07-21 by stale[bot]