dresden-elektronik/deconz-rest-plugin #1346: Change IP of DeCONZ to virtual interface

DurrM, 2019-03-11:
Hello,

so after I installed DeCONZ to my Raspberry I was no longer able to access the admin page of PiHole. I would like to keep PiHole on eth0 interface and "move" deconz to the virtual interface eth0:0. I'm just not sure if that's possible and how to do it.
Sorry but I'm pretty new. Hope somebody can help me or at least give me some link with further information.

Thank you

heitec11, 2019-03-11:
Both are using the port 80. So you have to change one of the programes to another port.

carsten-witt, 2019-03-12:
Change the port here: /etc/systemd/system/multi-user.target.wants/deconz.service

# sudo systemctl stop dezcon
# vi the file from above
# sudo systemctl daemon-reload
# sudo systemctl start deconz

Try with the new port

DurrM, 2019-03-12:
Worked like a charm. Thank you very much.

closed 2019-03-12 by DurrM