dresden-elektronik/deconz-rest-plugin #1350: [Request Device Support] Hive Door Sensor

jonb81, 2019-03-12:
Hello, 

Please would you consider supporting the Hive Window and Door Sensor.  I understand these used to be AlertMe sensors before the company was acquired by British Gas as part of their HiveHome ecosystem which is popular here in the UK but I'm not sure if they are available elsewhere.

They appear to be reachable through standard ZigBee HA clusters.  Interestingly they also appear to have an integrated temperature sensor.  Although this is readable it doesn't appear to be very accurate.

[image]

[image]

winterscar, 2019-05-13:
Bump for any of the Hive sensors or bulbs.

smithy2k3, 2019-06-02:
Bump, Could do with getting Hive Motion sensor working with Deconz
https://www.hivehome.com/products/hive-motion-sensor

smithy2k3, 2019-06-02:
`11:33:06:290 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:33:06:290 ZCL attribute report 0x00158D0003042024 for cluster 0x0400, ep 0x01
11:33:06:453 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
11:33:06:453 ZCL attribute report 0x00158D0003042024 for cluster 0x0406, ep 0x01
11:33:06:454 Beacon src: 0xDE97 ch: 25 updateId: 3
11:33:06:454 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:33:07:148 Set sensor check interval to 100 milliseconds
11:33:08:839 0x565F nwk changed to 0xE3AC
11:33:08:840 device announce 0x00124B0010B0ABAD (0xE3AC) mac capabilities 0x80
11:33:08:840 set fast probe address to 0x00124B0010B0ABAD (0xE3AC)
11:33:08:840 FP indication 0x0000 / 0x0013 (0x00124B0010B0ABAD / 0xE3AC)
11:33:08:840                       ...     (0x00124B0010B0ABAD / 0xE3AC)
11:33:08:840 device announce 0x00124B0010B0ABAD (0xE3AC) mac capabilities 0x80
11:33:09:860 [1] get node descriptor for 0x00124b0010b0abad
11:33:12:165 FP indication 0x0104 / 0x0500 (0x00124B0010B0ABAD / 0xE3AC)
11:33:12:165                       ...     (0x00124B0010B0ABAD / 0xE3AC)
11:33:12:248 wait response fastEnddeviceProbe() 0x00124B0010B0ABAD, elapsed 2388 ms
11:33:13:580 send permit join, duration: 59
11:33:14:325 FP indication 0x0104 / 0x0500 (0x00124B0010B0ABAD / 0xE3AC)
11:33:14:325                       ...     (0x00124B0010B0ABAD / 0xE3AC)
11:33:14:408 FP indication 0x0104 / 0x0500 (0x00124B0010B0ABAD / 0xE3AC)
11:33:14:408                       ...     (0x00124B0010B0ABAD / 0xE3AC)
11:33:14:409 wait response fastEnddeviceProbe() 0x00124B0010B0ABAD, elapsed 4548 ms
11:33:16:310 Beacon src: 0x91F4 ch: 25 updateId: 3
11:33:16:311 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:33:16:640 Beacon src: 0xDE97 ch: 25 updateId: 3
11:33:16:640 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:33:16:888 Beacon src: 0xB808 ch: 25 updateId: 3
11:33:16:888 Beacon src: 0xF553 ch: 25 updateId: 3
11:33:16:971 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:33:17:881 FP indication 0x0000 / 0x0013 (0x00124B0010B0ABAD / 0xE3AC)
11:33:17:881                       ...     (0x00124B0010B0ABAD / 0xE3AC)
11:33:17:881 device announce 0x00124B0010B0ABAD (0xE3AC) mac capabilities 0x80
11:33:17:881 FP indication 0x0000 / 0x0013 (0x00124B0010B0ABAD / 0xE3AC)
11:33:17:881                       ...     (0x00124B0010B0ABAD / 0xE3AC)
11:33:17:881 device announce 0x00124B0010B0ABAD (0xE3AC) mac capabilities 0x80
11:33:19:619 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:33:19:619 ZCL attribute report 0x00158D000302C743 for cluster 0x0400, ep 0x01
11:33:19:785 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
11:33:19:785 ZCL attribute report 0x00158D000302C743 for cluster 0x0406, ep 0x01
11:33:28:139 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:33:28:139 ZCL attribute report 0x00158D000302CC70 for cluster 0x0400, ep 0x01
11:33:28:471 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000`

smithy2k3, 2019-06-02:
11:34:12:550 send permit join, duration: 59
11:34:15:512 send permit join, duration: 0
11:34:16:011 Search sensors done
11:34:18:481 remove discovery request duplicate
11:34:18:677 don't close database yet, keep open for 900 seconds
11:34:22:817 0x00158D0001A12C7A level 255 --> 254
11:34:25:680 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:34:25:680 ZCL attribute report 0x00158D000302C743 for cluster 0x0400, ep 0x01
11:34:25:767 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
11:34:25:767 ZCL attribute report 0x00158D000302C743 for cluster 0x0406, ep 0x01
11:34:27:179 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:34:27:179 ZCL attribute report 0x00158D0003042024 for cluster 0x0400, ep 0x01
11:34:29:498 Incr. ZDP retry count 2 on item 7
11:34:36:648 sensor 18 (lumi.sensor_motion.aq2): disable presence
11:34:37:476 0x00158D00025BF17C error APSDE-DATA.confirm: 0xA7 on task
11:34:40:502 send permit join, duration: 59
11:34:43:196 Announced to internet
11:34:43:196 discovery server date: Sun, 02 Jun 2019 10:34:43 GMT
11:34:43:196 	 local time seems to be ok
11:34:43:278 Current channel 25
11:34:43:613 Device TTL 4380 s flags: 0x7
11:34:45:755 ZCL attribute report 0x000B3CFFFEF6D086 for cluster 0x0000, ep 0x01
11:34:47:726 dev /dev/ttyAMA0
11:34:47:888 GW firmware version: 0x26330500
11:34:47:888 GW firmware version is up to date: 0x26330500
11:34:58:478 sensor 5 (lumi.sensor_motion.aq2): disable presence
11:35:29:253 Beacon src: 0x9CD7 ch: 25 updateId: 3
11:35:29:253 Beacon src: 0x91F4 ch: 25 updateId: 3
11:35:29:253 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:35:30:426 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:35:31:649 Beacon src: 0xB808 ch: 25 updateId: 3
11:35:31:649 Beacon src: 0xF553 ch: 25 updateId: 3
11:35:31:649 Beacon src: 0x9CD7 ch: 25 updateId: 3
11:35:31:649 Beacon src: 0x3FC2 ch: 25 updateId: 3
11:35:33:849 0x4103 nwk changed to 0x896E
11:35:33:850 device announce 0x00124B0010B0ABAD (0x896E) mac capabilities 0x80
11:35:33:850 set fast probe address to 0x00124B0010B0ABAD (0x896E)
11:35:33:850 FP indication 0x0000 / 0x0013 (0x00124B0010B0ABAD / 0x896E)
11:35:33:850                       ...     (0x00124B0010B0ABAD / 0x896E)
11:35:33:850 device announce 0x00124B0010B0ABAD (0x896E) mac capabilities 0x80
11:35:34:841 [1] get node descriptor for 0x00124b0010b0abad
11:35:39:535 ZCL attribute report 0x00158D00025A5BB7 for cluster 0x0006, ep 0x01

smithy2k3, 2019-06-02:
[image]

smithy2k3, 2019-06-02:
[image]

pegatron89, 2019-08-13:
Did you manage to get this working on deconz? it connects like above through the gui but im unable to see it or use it on the webui

closed 2019-12-19 by stale[bot]

Puntoboy, 2020-02-05:
So I have been able to get my Hive bulbs to work with Deconz as they appear to be just rebranded Innr bulbs. When I try to pair the motion and door sensors I can see them in the main deconz window but they don't appear in the Phoscon webapp - thus are useless.

Puntoboy, 2020-12-06:
Any update on this?

Adjrobinson, 2021-06-24:
[image]
Bought one based on the above and it doesn't work.

Adjrobinson, 2021-06-25:
Sorry, I upgraded deconz and it works fine, apologies