dresden-elektronik/deconz-rest-plugin #1351: IKEA TRADFRI motion sensor, config attributes discrepancy?

lbschenkel, 2019-03-12:
Running deCONZ 2.05.60 with firmware 26320500.
This issue could have been present for a long time, I just happened to notice the discrepancy today when I was manually accessing the REST API.

I have a number of IKEA motion sensors, all of them running the same firmware (latest version, 1.2.214) and I noticed that all but one of them have both `duration` and `delay` attributes; the "odd" one has just `duration`, without `delay`.

```
    "11": {
        "config": {
            "alert": "none",
            "battery": 74,
            "duration": 10,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "[redacted]",
        "manufacturername": "IKEA of Sweden",
        "modelid": "TRADFRI motion sensor",
        "name": "motion_11",
        "state": {
            "lastupdated": "2019-03-12T15:38:25",
            "presence": false
        },
        "swversion": "1.2.214",
        "type": "ZHAPresence",
        "uniqueid": "00:0b:57:[redacted]-1000"
    },
    "13": {
        "config": {
            "alert": "none",
            "battery": 60,
            "delay": 60,
            "duration": 10,
            "group": "787",
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "[redacted]",
        "manufacturername": "IKEA of Sweden",
        "modelid": "TRADFRI motion sensor",
        "name": "motion_13",
        "state": {
            "dark": true,
            "lastupdated": "2019-03-12T19:54:55",
            "presence": false
        },
        "swversion": "1.2.214",
        "type": "ZHAPresence",
        "uniqueid": "00:0b:57:[redacted]-1000"
    },
   ...
   [omitting other sensors which are the same as 13]
```
I'm confused about two things:
1. How do identical devices, running the same firmware version, have different config attributes?
2. My understanding is that only the Philips Hue motion sensor should have `delay`, and IKEA sensors should only have `duration` (at least this is how I remember things) — how come `delay` showed up for these sensors? What is it controlling? Maybe it was after the firmware upgrade? (But then again, why not for sensor `11`?)

I was planning to delete and rejoin sensor `11` to see if the attributes change, but I decided to report it here instead so there's a chance to further diagnose this issue. @manup: any next steps you want me to take?

lbschenkel, 2019-03-12:
I have found https://github.com/dresden-elektronik/deconz-rest-plugin/pull/484 and https://github.com/dresden-elektronik/deconz-rest-plugin/issues/404#issuecomment-369028276 which shed some light regarding `delay` on IKEA motion sensors. If I understood it correctly `delay` is settable but overwritten with the value from the switch on the back of the sensor. It's only used as a fallback value when `duration` is set to 0. (Did I get this right, @ebaauw?)

So this leaves the question of why sensor `11` doesn't have a `delay`. I noticed it doesn't have a group either, and that might explain why `delay` attribute never got created (the links above mention the value being sniffed from group commands sent by the sensor). But don't all motion sensors have a built-in group? I can see the group corresponding to sensor `11` in the 'Wireless Light Control (2016)' UI — in fact there are 4 groups, one for each sensor (as I would expect).

ebaauw, 2019-03-12:
Yeah, the Trådfri motion sensor sends an _On with timed off_ command to its associated group, with the delay specified by the dial on the sensor.  The REST API plugin sets `config.delay` to the value in the command; it’s not settable through the API.

If you put lights in the group, controlling them directly from the sensor, without using the gateway, they’ll turn themselves off automatically after the timed off delay, without any further communication from the sensor.  I figured it would make sense for the sensor to report motion while the lights remain on.  That was before our understanding of the sensor needing time to “recharge” before it could send another command on detecting motion, which is handled through `config.duration`.

I think the REST API plugin only creates `config.group` when it sees the first group command.  I’m not sure when it creates `config.delay`: on pairing/database load or on receiving the command.  If both `config.group` and `config.delay` are missing, it could very well be the sensor hasn’t been bound to a group, and never sent a command.  Some types of sensors are bound to a random group on factory reset, others need to be bound explicitly, which the REST API plugin should do on pairing.

lbschenkel, 2019-03-13:
> Yeah, the Trådfri motion sensor sends an _On with timed off_ command to its associated group, with the delay specified by the dial on the sensor. The REST API plugin sets `config.delay` to the value in the command; it’s not settable through the API.

OK, good, so I got this right.

> If you put lights in the group, controlling them directly from the sensor, without using the gateway, they’ll turn themselves off automatically after the timed off delay, without any further communication from the sensor.

Yes, I'm aware. When I started used deCONZ (~1 year ago) I noticed that the old Wireless Light Control UI was showing groups that Phoscon wasn't, it's when I realized  that the sensors and remotes had their own groups and they could directly control devices that were put in there. I also noticed that the sensor itself is not a member of this group.

I never understood why those groups don't show up in Phoscon, though. (The groups for the remotes do, the groups for the motion sensors don't.)

> I figured it would make sense for the sensor to report motion while the lights remain on. That was before our understanding of the sensor needing time to “recharge” before it could send another command on detecting motion, which is handled through `config.duration`.

But today the implementation just uses `duration` (or `delay`) to reset presence after motion, not `duration+delay`, isn't it? Not that I'm complaining, I do rely on the fact that I can configure the sensor to reset the presence with a lower timeout than the IKEA minimum of 60s and I would be very sad if that went away. (I don't use the built-in groups because I have implemented more complex light automations based on state machines.)

> I think the REST API plugin only creates `config.group` when it sees the first group command. I’m not sure when it creates `config.delay`: on pairing/database load or on receiving the command. If [...]

Interesting that the group for that sensor *is* showing on the old Wireless Control UI (plus other 3 groups for the other sensors), and yet no `group` in the REST API. And in my case this sensors is in a high-traffic hallway, and it's guaranteed that I walked in front of it multiple times since last deCONZ restart. I know it reported motion (and sent the group 'on' command, I presume, since that's hardcoded firmware behaviour) since lights turned on every time I walked in front of it.

I did an experiment: I deleted all 4 groups in Wireless Control UI, refreshed, confirmed that they're gone, and walked in front of all 4 sensors. The groups reappeared in the UI, which make me conclude that deCONZ auto-creates the groups on demand when it sees the 'on with timed off' command sent to the group. I also noticed that `delay` and `group` then appeared for sensor `11` in the REST API. This is very odd, because as I said I'm positive that this sensor was being triggered frequently, and should have been sending the very same group commands, and deCONZ didn't react before but reacted now.

@manup: could you enlighten me about how the sensor/group mapping works at the Zigbee layer? Does the sensor independently choose a group to send group commands to, or is that set up by the gateway at join time? If the latter, how? It should be via something else then membership commands, since the sensors are not members of the group they send commands to (it's unclear to me if they even support being joined to a group since the group cluster is grayed out in deCONZ). Then what happens when the group is deleted in the UI? Is that purely on the UI or something happens in the Zigbee layer? Does the sensor start broadcasting commands to a new group or the previous group is simply rediscovered again when the sensor sends a command to it?

Since the "problem" went away I could close this issue, but it would be nice to understand why the group was not visible in the REST layer but visible in the Wireless Control UI, and it required a deletion of the group to fix it (sensor broadcasts were not sufficient). Is that expected somehow, or it could be a bug?

ebaauw, 2019-03-13:
> Does the sensor independently choose a group to send group commands to, or is that set up by the gateway at join time? If the latter, how?

Either, depending on the sensor type.

> It should be via something else then membership commands, since the sensors are not members of the group they send commands to

It’s a binding on the sensor from a _client_ cluster (typically _OnOff_ or _Level Control_) to the group.  You can create/delete these bindings yourself in the _Binding Dropbox_ panel of the deCONZ GUI.  Make sure to wake the sensor first.  Unfortunately, you cannot query them from the GUI.

> Then what happens when the group is deleted in the UI?

There’s no group object in Zigbee.  A group is more like a multi-cast address, to which a light subscribes, throught the _Groups_ server cluster.  Not sure, but ideally the REST API plugin should delete all the subscriptions of the member lights when the group resource is deleted.  The sensor is still bound to the group (or rather the address), so it happily continues to send commands to the group, causing the REST API plugin to re-create the group resource.

>  it would be nice to understand why the group was not visible in the REST layer but visible in the Wireless Control UI

The web app uses the REST API, so the group must be available there.  Like the Phoscon app, the web app uses a special request header to get slightly ritcher responses to some of the API calls (notably for the `/config` resource).  My best guess is that the group was hidden, so it wasn’t returned by the regular API.  My `ph` utility has an option to insert the special request header if you want to debug this.

lbschenkel, 2019-03-13:
Thanks for the answers. Just some tiny clarifications.

> > Does the sensor independently choose a group to send group commands to, or is that set up by the gateway at join time? If the latter, how?
> 
> Either, depending on the sensor type.

Appreciate it, but I'm referring specifically to IKEA sensors (and only IKEA sensors) in this issue. Do you know if for IKEA motion sensors, it's the former or the latter? 

> It’s a binding on the sensor from a _client_ cluster (typically _OnOff_ or _Level Control_) to the group. You can create/delete these bindings yourself in the _Binding Dropbox_ panel of the deCONZ [...]

In all of the IKEA motion sensors that I have, there's only the _OnOff_ cluster and it's greyed out. Cluster Info panel shows nothing. I'm fully aware that your knowledge about this is leaps and bounds above mine, but my understanding is that IKEA motion sensors don't support those clusters and are not configured in this way — they can only unconditionally send a "on with timed off" command to a group with a delay configured via the hardware switch. Am I completely missing the picture here?

> There’s no group object in Zigbee. A group is more like a multi-cast address, to which a light subscribes, throught the _Groups_ server cluster. Not sure, but ideally the REST API plugin should [...]

Understood.

> The web app uses the REST API, so the group must be available there. Like the Phoscon app, the web app uses a special request header to get slightly ritcher responses to some of the API calls [...]

I want to clarify that I was talking about 2 things, and I think they got mixed up:
1. The `group` (and `delay`) attribute in `/sensors/11/config`, which was not present for 1 out of 4 identical sensors until I have deleted the group via old web UI (Wireless Control) even after walking in front of the sensor many times and knowing for a fact that the sensor reacted. It looks like it might be a bug.
2. The groups corresponding to the motion sensors that show up in the old web UI but not Phoscon. Not really a problem there but I want to understand better why they show up in one and not the other. Probably a deliberate decision, I just want to understand the reasoning.

Regarding 1, that is was I was originally curious about and the subject of this issue.

Regarding 2, it was brought up later after your explanation. These groups always showed up in `/groups`. Curiously enough, the sensor is inside `devicemembership` (even though the device is not really a member). But they *never* showed up in Phoscon, only the old web UI, and I'm 200% sure about this, since the very first version I started using a year ago. Since that time the network has been recreated, these sensors have been reset and rejoined more than once and their firmware was updated. And yet the groups never showed in Phoscon. (To be clear: the groups for the remotes do show, but not the groups for the motion sensors.)

> My `ph` utility has an option to insert the special request header if you want to debug this.

Yes, please. Or if it's just the matter of adding a hardcoded header, just tell me the name/value so I can add to the `curl` invocation.

ebaauw, 2019-03-13:
> Do you know if for IKEA motion sensors, it's the former or the latter?

I'm afraid I don't.  I would have to reset and re-pair mine while sniffing to find out.

> there's only the OnOff cluster and it's greyed out

Clusters come in pairs: a server (blue number) and a client (grey number).  In my experience, the server is the one that implements state, but there's probably some exceptions to that.  Request commands are sent from a client cluster to the corresponding server cluster; responses and notifications from a server cluster to the corresponding client cluster.  So the _OnOff_ server cluster provides the _OnOff_ attribute and receives _On_ and _Off_ commands; the client _OnOff_ server sends _On_ and _Off_ commands.

`ph` is bundled with homebridge-hue, but you can also install the utilities standalone, see https://github.com/ebaauw/homebridge-hue-utils.  Functionally, it combines curl and a json formatter, providing a more accessible syntax (see `ph -h`).  

To see the difference for groups (note the `hidden` attribute):
```
$ ph get /groups/1
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [],
  "etag": "ab3676fa83f1517fdf6a5c732a236a6a",
  "id": "1",
  "lights": [
    "241",
    "242",
    "324",
    "281",
    "282",
    "342",
    "332",
    "234"
  ],
  "name": "On after power cycle",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
$ ph -p get /groups/1
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [],
  "etag": "ab3676fa83f1517fdf6a5c732a236a6a",
  "hidden": false,
  "id": "1",
  "lights": [
    "241",
    "242",
    "324",
    "281",
    "282",
    "342",
    "332",
    "234"
  ],
  "lightsequence": [],
  "multideviceids": [],
  "name": "On after power cycle",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```

> Or if it's just the matter of adding a hardcoded header, just tell me the name/value so I can add to the curl invocation.

I think the header is:
```
Accept: application/vnd.ddel.v1
```

lbschenkel, 2019-03-16:
> I think the header is:
> 
> ```
> Accept: application/vnd.ddel.v1
> ```

Yep, that's it. It worked when I tried it.

@manup: what would be the reasoning to hide the associated groups for the motion sensors from the Phoscon UI (and not the old web UI)? I searched the source code to see if I could figure it out but it looks like that the logic to set `hidden=true` for the group is implemented in deCONZ, not the REST plugin. At least I wasn't able to find it.

ebaauw, 2019-03-16:
> the logic to set `hidden=true` for the group

I think it's in the old web app, together with the sorting of lights within a group, and the sorting of groups for display in the groups page.  The deCONZ core doesn't know about any REST resource.

closed 2019-07-21 by stale[bot]