dresden-elektronik/deconz-rest-plugin #1353: .60 safe to try?

olemr, 2019-03-13:
It's been 3 days without any release notes.
Anyone tried it and the new Firmware?

bdickhaus, 2019-03-14:
I am running it on three Raspbee's with no trouble so far. I have Hue bulbs, switches and sensors, Xiaomi sensors and buttons, and a few other odds and ends.

olemr, 2019-03-14:
Thanks.
Now I'm running it too.
[image]
Hope Ikea bulb handling has improved :-)

MattL0, 2019-03-15:
@olemr may I asked why you have 3 bridges?

is it for performance? or distance?

Thanks. 

Now, I have somewhat 50 zigbee devices, so I am wondering about the future :P

bdickhaus, 2019-03-15:
I'm the one with three bridges. I have two at one location, one is the primary bridge, with about 50 devices, and the other has one bulb and a few devices for testing. The third bridge is at another location and I am just getting it set up.

olemr, 2019-03-15:
I too have 3 bridges, but only one Conbee. The other 2 are Ikea and HUE.
Seeking out the best.
I integrate with openHab and the binding for Ikea works fine, but is lacking on group operations. Native Zigbee groups are not supported and a group of say 12 GU10 bulbs use 2-3s to settle.
Then i use Conbee/deconz to interface my mi cubes, and transferred som 30 Ikea bulbs as well. Group operations are synchronous, but bulbs often becomes unreachable and has to be power cycled.
Hence i recently added a HUE bridge and transferred 6 GU10 bulbs to that one. REST API to HUE is identical to deconz, so now I'm doing some long term testing.
Exiting how .60 fares with Ikea bulbs. If they are still lost, I will try adding more of them, to HUE.

olemr, 2019-03-22:
Update: still losing Ikea bulbs. 
Will try HUE gateway next.

manup, 2019-03-22:
>Update: still losing Ikea bulbs.
>Will try HUE gateway next.

If possible can you try to recreate the same setup the Hue bridge, especially same amount of IKEA lights. 

There is ongoing work to fix issues with IKEA lights but since deCONZ handling is now very close to the "original" IKEA gateway handling I'm not sure lost lights can be prevented totally from the gateway or if there is a deeper problem within the lights firmware, which can only be fixed in future firmware releases.

So please report back if your Hue gateway manages to hold your full setup of 30+ Ikea lights online and stable (can only be verified by control lights individually). We would have a more clear picture if the problem can be fixed at the gateway side.

olemr, 2019-03-22:
> So please report back if your Hue gateway manages to hold your full setup of 30+ Ikea lights online and stable 

Will do. Other sensors and lights are stable, so I won't be leaving completely :-)

djwlindenaar, 2019-03-23:
I'm on .60 now for 1.5 weeks. Yesterday I noticed one of the lights did not turn on (group command based on rule). 
In the UI it still shows yellow, but no connection lines anymore. It shows unreachable in REST. Pressing F5 did not reconnect it.

Anything I should try logging or otherwise that may help?

manup, 2019-03-23:
You may try selecting the light and press 0, but since it doesn't react on group commands I'm afraid it won't come back. Therefore likely power-cycle of the light is needed.

How many devices do you have in your network?

djwlindenaar, 2019-03-23:
It says 53 nodes. 
25 IKEA lights
1 jiawen light strip
3 Xiaomi plugs
5 Xiaomi switches
8 Xiaomi temp sensors
2 IKEA motion sensors
3 Xiaomi motion sensors

QonosSWE, 2019-03-25:
hmm how do i install this on my rpi? hehe i cant figure that out. ok im new to this, but i thought it would auto update to 2.05.60, but im still on .59 and well.. can someone please point me in right direction?

Smanar, 2019-03-25:
To install the .60 version ?
You have all version for Pi here > https://www.dresden-elektronik.de/deconz/raspbian/beta/
On my side I just install the new one on the older one.

QonosSWE, 2019-03-25:
yeah i tried to do that on my rpi, but its says  command not found when i try to use dpkg same with "apt -v"

i installed after this guide:
https://www.home-assistant.io/getting-started/

and i installed "SSH & Web Terminal", but still cant use them when i try to run them with protection mode disable.

bdickhaus, 2019-03-25:
If you are running hass.io then you are running an addon and can't update deconz directly. You need to update the hass.io deconz addon. This one, which is probably the one you are using, has been updated to .60. [https://github.com/marthoc/hassio-addons](url).

QonosSWE, 2019-03-25:
Ok, finaly things start to make sense.. hehe spent alot of hour trying to find a way to get ite work, but i was doing it the wrong whey then..
I use:
https://www.home-assistant.io/components/deconz/
so maybe i should give the other one a try since this one i have still say .59

olemr, 2019-03-26:
Test on HUE started.
28 Ikea Bulbs. 16 E27 and 12 GU10 ct bulbs.
Online since yesterday evening.
[image]

Transfer went smooth even though bulbs are up to 10m away.
Procedure: 6x power cycle - reset, power off, start scan in app, power up.

manup, 2019-03-27:
Cool I'm very interested in results. If possible check from time to time that all lights are controllable, per single device. To my knowledge the Hue bridge does also heavy polling but only state attributes of the lights. Also the bridge doesn't configure any attribute reporting or queries of neighbor tables.

If this still works reliable with the lights after ~2 weeks we can try to adapt.

Are these lights always powered or sometimes powered off?

olemr, 2019-03-27:
> Are these lights always powered or sometimes powered off?
Always powered.
Adapting REST API was a breeze.
Dusk animation goes on for 80mins, with brightness from 0%->80% every minute and CT 80%->0% 30s after brightness. These are all group commands. Will try individual bulbs as well.

djwlindenaar, 2019-04-01:
Again one of my IKEA lights went missing. A different one from last time.

Pressing 0 didn't do it. Power cycle did.

djwlindenaar, 2019-04-02:
@manup, interesting one, maybe. The light that disconnected was power cycled a while ago together with two others. (Fuse blown)

Less than a day after the first, another went missing. Could this indeed indicate some systematic thing?

olemr, 2019-04-04:
Well, what do you know.
Today I noticed a hung bulb on HUE as well:
[image]

So, same issue on both systems.

manup, 2019-04-04:
Yes, it looks like something brings the light firmware into a bad state.

I have currently three dead lights which I'm looking at in the sniffer, see: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

I'm not sure if that can be solved/worked around at the gateway level.

closed 2019-08-09 by stale[bot]