dresden-elektronik/deconz-rest-plugin #1354: Xiaomi vibration sensor; information leckage between deconz and Phoscon

heitec11, 2019-03-14:
Phoscon is only update the status with an vibration.
After 2 days the acuall status is not reachable.
[image]

But deconz shows that the conection is still alive:
[image]

The value "lastupdated" shows the last "buttonevent".
Maybe 2 values are nessary. "lastupdated_buttonevent" and "lastupdated"

MattL0, 2019-03-15:
I have 6 of those. But just added one to deconz. Most of the time it is showing that is is not reachable

jerrychong25, 2019-03-26:
Similar issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1378

closed 2019-07-31 by stale[bot]