dresden-elektronik/deconz-rest-plugin #1356: Phoscon doesn't show the new firmware version

Mr-REM-Esq, 2019-03-15:
[image]

OakNinja, 2019-03-16:
It's just a time stamp issue, previous version was 2.05.59 / 14.02.2019.

Ronoon, 2019-03-16:
The Firmware update was show after powercycle the raspberry 
Maybe you should try....

Mr-REM-Esq, 2019-03-16:
> The Firmware update was show after powercycle the raspberry
> Maybe you should try....

It's not a raspberry. It's Debian. I restarted the system many times, but there are no differences.

Mr-REM-Esq, 2019-03-16:
> EDIT:
> 
> Missed the firmware part, do you run on Windows?

I managed to update the firmware under Windows. Nevertheless, it seems to me that there is an error in the driver.

Nosion, 2019-03-17:
Had the same behavior!
Setup:
Ubuntu 18.10, with KDE desktop (connected to TV)
Normally I run deconz as a service.

1. - Stopped the deconz service
2. - Started the Desktop GUI app
3. - Wend to the pwa web-site ->gateway, got the upgrade firmware button.
4. - Updated firmware
5. - Closed the desktop app and started the service again.

The systemd deconz config:
```
[Service]
ExecStart=
ExecStart=/usr/bin/deCONZ -platform minimal --ws-port=8003 --http-port=8001 --auto-connect=1 --dbg-error=1
```

manup, 2019-03-17:
All versions up to the current deCONZ version will only show the firmware update button if only one (FTDI based) USB-dongle is plugged in.

As workaround the firmware can be updated manually via GCFFlasher as described in the wiki:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

In future deCONZ versions the updated will handle multiple USB devices better to pin the update to the device which is actually connected.

Mr-REM-Esq, 2019-03-18:
> All versions up to the current deCONZ version will only show the firmware update button if only one (FTDI based) USB-dongle is plugged in.
>
> As workaround the firmware can be updated manually via [...]

But I have only one USB-dongle.

manup, 2019-03-18:
Hmm can you post the output of:

    sudo GCFFlasher_internal -l

Mr-REM-Esq, 2019-03-18:
> Hmm can you post the output of:
> sudo GCFFlasher_internal -l

Sure.

ftdi_sio
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10

       1  FTDI device found

 device | vendor | product | serial  | description
--------|--------|---------|---------|----------------------
failed to open FTDI device (0) status = 3

manup, 2019-03-18:
Indeed looks like just one device. I don't yet know why the update button isn't shown but here here is the command to manually update the firmware.

1. close deCONZ
2. exec following command

`sudo GCFFlasher_internal -d 0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26320500.bin.GCF`

Mr-REM-Esq, 2019-03-18:
> here here is the command to manually update the firmware

Thank you for your help, but I already solved the problem. I pulled out the USB-dongle from the computer with Debian and inserted it into the computer with Windows. The latest version of deCONZ was preinstalled on the computer with Windows. The button appeared and everything went smoothly. Then I returned the dongle to the computer with Debian.
Now it looks like this:
[image]

Do you see the time stamp? On Debian this stamp is 14.02.2019, but on Windows the same dongle shows 10.03.2019. It's strange.

manup, 2019-03-18:
> It's strange.

That's the wonderful world of browser cache :)
It should refresh after a while.

closed 2019-07-23 by stale[bot]

ma-zal, 2021-12-10:
Issue is still unresolved. Please reopen ticket.