dresden-elektronik/deconz-rest-plugin #1358: Missing battery level in Aqara Multisensor WSDCGQ11LM (2.05.60)

Mr-REM-Esq, 2019-03-15:
I installed the new firmware version and readded the sensors.
[image]
[image]

QonosSWE, 2019-03-26:
[image]

Same here, but the round sensor have battery status.

closed 2019-04-16 by Mr-REM-Esq

Smanar, 2019-04-19:
Hello, just by curiosity, the problem was resolved by itself, or you have done something ?
I know someone with the same problem, all devices are goods except this one (and 2 time the same device).

Mr-REM-Esq, 2019-04-20:
I completely deleted the old version of deconz and installed the new one (2.05.63).

[image]

[image]

Smanar, 2019-04-20:
I know someone with this version too, but same bug
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1450

Perhaps the same problem than your problem update from the sensor that did not come within 3 hours.

Mr-REM-Esq, 2019-04-20:
Yes, it looks probably.

bcutter, 2021-01-24:
What was the "solution" for this?