dresden-elektronik/deconz-rest-plugin #1359: Unable to start Phoscon in .60

mk1blacklimited, 2019-03-16:
I get this with my RaspBee no matter what browser I try (tried Safari, Chrome, Edge):

This Browser is not supported :-(
Please use one of the supported browsers:

Chrome
Firefox
Safari
Samsung Browser
Edge
This browser identifies itself as:
Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_4) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/12.1 Safari/605.1.15

manup, 2019-03-16:
Looks like an Javascript error is thrown.

Can you please open developer console (F12) and reload the page.
There should be a a line in the console starting with `Error: ...`

mk1blacklimited, 2019-03-16:
This is what I get in the console:

Error: Uncaught SyntaxError: Invalid or unexpected token
languages.js?231bcd:90 Uncaught SyntaxError: Invalid or unexpected token
login.js?40509e:3 Use of the Application Cache is deprecated on insecure origins. Support will be removed in the future. You should consider switching your application to a secure origin, such as HTTPS. See https://goo.gl/rStTGz for more details.
jquery-min.js?0baf29:2 jQuery.Deferred exception: loadLanguage is not defined ReferenceError: loadLanguage is not defined
    at HTMLDocument.$.ready (http://192.168.1.71/pwa/js/login.js?40509e:3:9131)
    at j (http://192.168.1.71/pwa/library/js/jquery-min.js?0baf29:2:29999)
    at k (http://192.168.1.71/pwa/library/js/jquery-min.js?0baf29:2:30313) undefined
undefined:1 Uncaught (in promise) TypeError: Illegal invocation
login.html:615 Error: Uncaught ReferenceError: loadLanguage is not defined
jquery-min.js?0baf29:2 Uncaught ReferenceError: loadLanguage is not defined$.ready @ login.js?40509e:3j @ jquery-min.js?0baf29:2k @ jquery-min.js?0baf29:2
blob:http%3A//192.168.1.71/07f5de25-a989-4831-b1b0-656a171d30dc:101 scanner worker init
blob:http%3A//192.168.1.71/07f5de25-a989-4831-b1b0-656a171d30dc:333 add ip 192.168.1.71
blob:http%3A//192.168.1.71/07f5de25-a989-4831-b1b0-656a171d30dc:333 add ip 192.168.1.130
blob:http%3A//192.168.1.71/07f5de25-a989-4831-b1b0-656a171d30dc:335 already know ip 192.168.1.71
blob:http%3A//192.168.1.71/07f5de25-a989-4831-b1b0-656a171d30dc:236 found gw: 192.168.1.71 
jquery-min.js?0baf29:4XMLHttpRequest cannot load http://192.168.1.71:undefined/api/8D2656F7A1?_=1552737610436. Cross origin requests are only supported for protocol schemes: http, data, chrome, chrome-extension, https, chrome-extension-resource.

manup, 2019-03-16:
That's odd the line 90 in languages.js looks normally, not sure why it throws a `SyntaxError: Invalid or unexpected token`.

Just a wild guess, but please try reinstall 2.05.60, maybe the file is corrupted somehow – like due sd-card errors.

mk1blacklimited, 2019-03-16:
Reinstall didn't fix it unfortunately. Is it possible to revert back to .59?

mk1blacklimited, 2019-03-16:
Huh, started an Incognito window in Chrome, and logged in to Phoscon as normal, everything is working there.

Still no go through my usual browsers though.

OakNinja, 2019-03-16:
> Huh, started an Incognito window in Chrome, and logged in to Phoscon as normal, everything is working there.
> 
> Still no go through my usual browsers though.

It's a local browser cache issue. In chrome, open developer tools, select the Network tab, tick "Disable cache" and reload the page and it will probably work.

mk1blacklimited, 2019-03-19:
I can understand ONE browser, but I've tested with a lot on different computers (and an iPhone), and not one of them works. I've probably had Phoscon running on all of them before, but still...:) Anyway, not a huge problem.

closed 2019-07-24 by stale[bot]

vjonas, 2023-03-25:
I have the same CORS issue.
When it tries to fetch the information from the correct homebridge ip settings.

Mimiix, 2023-03-25:
Hi

If you have any issues, please reside to the correct help channels.

I'm locking this as its a 3 year old issue.

Kind regards,
Dennis