dresden-elektronik/deconz-rest-plugin #1360: Toogle doesnt work with multiple wall switches

thewhobox, 2019-03-16:
If you have more than one wall switch deconz doesnt toggle the Group state, it toggles the switch state.

For example:
If you turn a Group on with switch 1 (toggle Event) and turn it off with Switch 2 (only off Event, or with alexa or wathever) you will have to press twice on switch 1 to get it back on.
(Also when you want to turn it off)

It doenst matter if switch 2 toggles too.

daaaaa, 2019-03-17:
I confirm this behavior. The same happens to me in the very same situation.

etique57, 2019-03-17:
I confirm this behavior too.

ademobile, 2019-03-18:
Yes, they seem to use a flag to track the light state rather than check if any light is on. This my biggest reason why I do not use the switch editor at the moment. I ending up picking up the event from the REST API, querying the state of the light in question and NOT(ting) the any_on state

daaaaa, 2019-03-21:
This is a huge problem, imho, and should be resolved. It makes very boring the system becuase i usually have to press a switch twice to turn off lights when I go to sleep :p

DOliana, 2019-04-01:
I can confirm the issue. There was one verion of deCONZ which did not have this problem - sadly I cannot remember which one it was (I am talking a few months ago though).

thewhobox, 2019-04-09:
Any news or comment from a developer?

PhiHar, 2019-04-14:
I also can confirm the issue. When going to sleep and my girlfriend turned the light with switch1 on, i have to press switch2 twice to turn it off.

daaaaa, 2019-04-18:
> Any news or comment from a developer?

I'd like to have news about this issue since this behavior is quite annoying.

capstan1, 2019-07-12:
I just noticed exactly the same thing. Is there anything we can do about it?

DOliana, 2019-08-30:
has been fixed in 2.05.67 (see [1181](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1181#issuecomment-525224516) @manup I guess you can close this issue.

HackTik, 2019-08-30:
amazing, then i have to update today evening and testing this out. It was one really annoying issue for me aswell.

closed 2019-09-02 by thewhobox