dresden-elektronik/deconz-rest-plugin #1362: Request adding of vibration state/attributes for Xiaomi Aqara Vibration Sensor in Home Assistant

jerrychong25, 2019-03-19:
Hi all,

I have added **Xiaomi Aqara Vibration Sensor** via deCONZ integration plugin into Home Assistant.

However, it show only **sensor.vibration_sensor_battery_level** for current entities, as per screenshot below:
[image]

So, I can't access this sensor vibration state/attributes due to this problem.

Could deCONZ add this vibration sensor state/attributes into Home Assistant?

So that I could display this vibration sensor state/attributes into Home Assistant Home Page.

Thanks.

WhistleMaster, 2019-03-21:
Have a look here [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/748](url)

jerrychong25, 2019-03-22:
> Have a look at #748

Looks like it does not have this option as mentioned in
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/748#issuecomment-439899605:
[image]

Possible have any changes by adding this vibration sensor state/attributes into HA soon?

imbaimbaimba, 2019-03-22:
> > Have a look at #748
> 
> Looks like it does not have this option as mentioned in
> [#748 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/748#issuecomment-439899605):
> [image]
> 
> Possible have any changes by adding this vibration sensor state/attributes into HA soon?

I'm also interested in getting it into Home Assistant. No clue whether it is possible or not?
Can we have an update on this? :)

Kane610, 2019-03-25:
@imbaimbaimba see my latest comment in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/748#issuecomment-476242634

jerrychong25, 2019-03-27:
@Kane610,

Thanks for the updates.

So this fix will only appear on 0.90.2?

Kane610, 2019-03-27:
> @Kane610,
> 
> Thanks for the updates.
> 
> So this fix will only appear on HA 0.90.2?

New features are not added in betas. 0.91

jerrychong25, 2019-03-28:
> > @Kane610,
> > Thanks for the updates.
> > So this fix will only appear on HA 0.90.2?
> 
> New features are not added in betas. 0.91

Noted and thanks

jerrychong25, 2019-06-24:
Since this issue no longer affect my Raspberry Pi performance, I will close this issue.

closed 2019-06-24 by jerrychong25