dresden-elektronik/deconz-rest-plugin #1364: Phoscon scene dont change color temperature (tradfri)

tscheerio, 2019-03-19:
I saw the Bugfix in the color temperature,
but the Scenes are still not working properly.

I created two scenes, one with full brightness and full warm light and the other one with full brightness and full cool light.
When i manually push the slider up and down the color temperature is changing to the right state but when i use the defined scenes to change the color temperature nothing is happening, i always get a "normal" colored light.

I use the IKEA Tradfri GU10 WS 400lm
Thought this was fixed.
Whats wrong here?

lbschenkel, 2019-03-19:
Are you configuring the scene via group commands by any chance? In other words, are you setting it up in Phoscon by having the selector under "Light control" set to "all"?

IKEA lights have a firmware bug in which they don't properly save the color temperature when storing scenes. deCONZ has implemented a workaround for it but **only** for unicast commands (sent directly to the light). You'll have to change that switch to "single" and change the color temperature (to make sure the unicast command is sent), and then you immediately save the scene. This should work.

Sometimes I also forget about this and get bit by it too. It's a very annoying bug in the IKEA lights.

tscheerio, 2019-03-19:
Ok i got it.
I was configuring them via the group command, didnt know that an easy switch to the single commands will solve this problem. 
ty :)
Case closed i think^^

realwax, 2019-04-01:
I chased this through and could solve it via old web app. Corrected the lamps with wrong settings and stored the scene there and it works. I think it is a deconz/phoscon bug...

johnssson, 2019-04-11:
Where can i see the firmware version?
Is it this?
Hersteller: IKEA of Sweden
Modell: TRÅDFRI Lampe E27 WS opal 980lm
Version: 1.2.217
I have the Bug via Phoscon WebApp

closed 2019-08-16 by stale[bot]

E1onE2, 2020-10-22:
I got the same issue with Phoscon and a smart light strip controller (RGBCW).
Manually I can choose to use color temperature, but it is not saved in the scene.

rovingrambler, 2020-11-14:
same issue with a segled element
scene only saves brightness