dresden-elektronik/deconz-rest-plugin #1365: Xiaomi Magic Cube - Rotate events for Home Assistant / Node Red

Cavemanz, 2019-03-19:
Hey

Ive managed to set up all the relevant event IDs in Node Red to catch the regular actions, flip90/flip180/shake/push/double tap.

However, I cant seem to find a good way to handle rotate.
Any suggestions on how to go about catching all the enormous amounts of different IDs it can return? Is it possible to send 2 events? Like 7010 if turning clockwise, 7011 if turning counter clockwise?

Regards
Morten

Cavemanz, 2019-03-20:
I managed to sort the events through a filter node with the help of @kermit from the HA Discord. For anyone else with a similar problem, this is my Node Red Function to catch rotates:

``const code = msg.payload.event.event;
let action;

const codes = { 
    "side1": [1002, 1003, 1004, 1005, 1006],
    "side2": [2001, 2003, 3004, 4005, 5006],
    "side3": [3001, 3002, 3004, 3005, 3006],
    "side4": [4001, 4002, 4003, 4005, 4006],
    "side5": [5001, 5002, 5003, 5004, 5006],
    "side6": [6001, 6002, 6003, 6004, 6005],
    "push": [1000, 2000, 3000, 4000, 5000, 6000],
    "doubletap": [1001, 2002, 3003, 4004, 5005, 6006],
    "shake": [7007],
    "drop": [7008],
    "wake": [7000]
};

for (let a in codes) {
    if(codes[a].includes(code)) {
        action = a;
        break;
    }
}

if (!action) {
    const strCode = String(code);
    if (strCode.substr(1,2) !== "00" || strCode.length !== 4) {
        action = code > 0 ? "rotatecw" : "rotateccw";
    }
}

msg.payload = action || "";

return msg;``

ebaauw, 2019-03-20:
There’s actually two ZHASwitch resources for the cube: one for the digital events (push, flip, drop, etc), and one for the analogue events (rotation).  The digital resource returns the event codes; the analogue resource returns the rotation angle.  You don’t need to check for a known digital code, when you know the originating resource.

Snailboy2, 2019-03-20:
I'm using this and it works great. 
https://community.home-assistant.io/t/xiaomi-cube-rotate/34186/37

Hiffi, 2019-04-20:
For me, the analogue values "03-000c" are not accessible via the REST API. Can I change this somehow?

Cheaterdev, 2019-06-06:
The problem that in Home Assistant we can't determine where it is a digital event or analog event.
With checking by known states there is possibility to turn cube for a angle that will be detected as digital event.

Example of digital event:
`{
    "event_type": "deconz_event",
    "data": {
        "id": "cube",
        "event": 1000
    },
    "origin": "REMOTE",
    "time_fired": "2019-06-06T22:54:10.505890+00:00",
    "context": {
        "id": "53340fe63016439aa60112b478da305e",
        "parent_id": null,
        "user_id": null
    }
}`

Example of analog event:
`{
    "event_type": "deconz_event",
    "data": {
        "id": "cube",
        "event": 10378
    },
    "origin": "REMOTE",
    "time_fired": "2019-06-06T22:56:53.710696+00:00",
    "context": {
        "id": "91c0405fd0004ee7a7f9bad27e27e2dc",
        "parent_id": null,
        "user_id": null
    }
}`

Lefey, 2019-06-23:
> The problem that in Home Assistant we can't determine where it is a digital event or analog event.
> With checking by known states there is possibility to turn cube for a angle that will be detected as digital event.

It is because you has rename cube in phoscon web ui, if you delete it and add again, in home assistant there two event_id such as switch_1 and switch_2.
Added this as issue  #1624

closed 2019-10-28 by stale[bot]