dresden-elektronik/deconz-rest-plugin #1366: aqara pir with long holdtime

thrusty66, 2019-03-20:
I control the light some of my Ikea lamps via [aqara pir](https://de.aliexpress.com/item/Original-Xiaomi-Aqara-Mijia-Smart-Temperatur-Menschlichen-K-rper-Sensor-PIR-Motion-Sensor-Bewegung-Home-Security/32846970782.html).

Unfortunately, this Pir has the habit of saving the last movement for some time. 
If you have come into the area of the pir's, then the activity will continue for a long time, although nothing more can be seen for the pir. 
Is there a way via Phoscon to change this hold time? Or is the problem due to the firmware of the aqara pir?

WhistleMaster, 2019-03-21:
You can change the "duration" parameter in the configuration of the sensor using curl:

`curl -H 'Content-Type: application/json' -X PUT -d '{"duration": 1}' http://DECONZ_IP/api/API_KEY/sensors/SENSOR_ID/config`

thrusty66, 2019-03-23:
Thanks for the answer.
Unfortunately I get an error when I enter the Curl command in the console.
I use the Sensor_ID and API_Key which pimatic took over. 
I'm getting this issue:
`[{"error":{"address":"/sensors/00158d0002732896","description":"resource, /sensors/00158d0002732896, not available","type":3}}`

WhistleMaster, 2019-03-23:
You have to grab your SENSOR_ID from the Rest API: `http://DECONZ_IP/api/API_KEY/sensors/` and your IDs should only be numerical, without any letters, and also little numbers (e.g. 12,3,5,...)

thrusty66, 2019-03-23:
I'm doing something wrong.
`ingo@debian:~$ http://192.168.178.32/api/XXXXXXXXXX/sensors/`
`
-bash: http://192.168.178.32/api/XXXXXXXXXX/sensors/: Datei oder Verzeichnis nicht gefunden`

WhistleMaster, 2019-03-23:
Just open the URL in your web browser and you will see all the sensors. Just find the Xiaomi Motion you would like to configure, and then, use the curl command to modify the "duration".

kugelkopf123, 2019-03-23:
> I'm doing something wrong.
> `ingo@debian:~$ http://192.168.178.32/api/XXXXXXXXXX/sensors/`
> ` -bash: http://192.168.178.32/api/XXXXXXXXXX/sensors/: Datei oder Verzeichnis nicht gefunden`
> 
> Edit:
> I'm sorry,
> I have now entered this in the browser and received an output.

Don't forget the Port! its http://IP:PORT/api/API_KEY/sensors/ Normaly its Port 8080

thrusty66, 2019-03-25:
Hello, now I have managed to set the time shorter.
Unfortunately this creates a next problem. A certain time after the PIR has reported a movement he does not react. That means after about 50sec he reacts to a movement again. 
Is there also a setting possibility for this?

kugelkopf123, 2019-03-25:
> Hello, now I have managed to set the time shorter.
> Unfortunately this creates a next problem. A certain time after the PIR has reported a movement he does not react. That means after about 50sec he reacts to a movement again.
> Is there also a setting possibility for this?

No =(! The AQARA PIR Sensor sleeps for 60 sec. after it got triggerd. 60sec later it is able to get triggerd again. The duration you had set, discribes only the time how long the presence value stays true. On default it is set to 90 because, if the PIR Sensor gets retriggerd after the 60sec delay from the PIR Sensor itselfs, the presence Value stays true.

thrusty66, 2019-03-25:
Okay, thanks for the info. So, if I have the time > 60 (> 90), then it triggers when there is still movement. If the movement is before the time, then he does nothing.
It's still stupid, but what can't be done, can't be done. Too bad.

closed 2019-07-31 by stale[bot]