dresden-elektronik/deconz-rest-plugin #1367: OSRAM Lightify Lights (US ZHA Version) - Missing Color Temperature

stephenfsnow, 2019-03-20:
Greetings! I've been trying out deconz and have noticed that the color temperature selector is missing in the Phoscon web app for all my RGBW lights that support color temperature.

Some info about the Sylvania OSRAM lights that are missing color temperature configuration. All are the US, ZHA variety...
OSRAM - LIGHTIFY RT RGBW - FW Version 01020492
OSRAM - LIGHTIFY Flex RGBW - FW Version 01020492
OSRAM - LIGHTIFY Outdoor RGBW - FW Version 01020492

I also have some A19 RGBW lights that are Sylvania branded but were made by LEDVANCE. They are the only bulbs that I'm able to change color temperature.
LEDVANCE - A19 RGBW - FW Version 00102411

Please let me know what information I can provide to help with this issue. I'd love to use deconz to combine all my Zigbee bulbs to one hub but since I rely heavily on color temperature settings, I'm unable to do that as of now.

Thank you!!

stephenfsnow, 2019-03-27:
I wonder if the plugin is treating my OSRAM Lightify bulbs as ZLL type instead of ZHA? Is there a way to force ZHA? Just shooting in the dark here since color temperature works great with the ZHA component in Home Assistant using the RaspBee board. However if I switch to the deCONZ component that uses this plugin, the OSRAM Lightify bulbs are way off when setting color temperature.

bdickhaus, 2019-05-03:
I am having what I think is a related problem with a Lightify A19 Tunable White (US version). I took a quick look at the deCONZ code. Some (all?) Lightify bulbs do not support the color capabilities attribute. At this point deCONZ hard codes capabilities of hue and sat . Note that ct is missing. Interestingly enough, I can send a "put" with a ct value and it works. But the color temperature value in the color control cluster was 0. Until I read it manually. Now it is updating, and is now showing up in the REST api response (before I did the manual read it was not showing up). Phoscon thinks it is a color bulb. It doesn't respond to hue/sat but the ct does work. I am not positive that reading the color temperature value in the color control cluster is what got it to start working, but you night try that.

HomeAssistant has a number of things hardcoded specifically for Lightify bulbs.

closed 2019-09-07 by stale[bot]

DiggaTS, 2021-07-13:
2 years later and the issue continues ...