dresden-elektronik/deconz-rest-plugin #1368: Password STILL not being remembered.

ademobile, 2019-03-20:
I am still having issues with password not being remembered (using Safari on Mac or iOS). I do not have any privacy modes switched on, or cache clearing etc. I just have the standard 'out the box' settings.

I have asked this before and I will ask again, why do we need a password just to turn lights on or off via the web GUI. You do not password protect a wall switch!

Please either gives us the option to disable password, or better still only password protect changes to the system, like adding, removing or changing resources, just not basic day to day functions!

manup, 2019-03-21:
>I have asked this before and I will ask again

What was the related issue?

Can we please keep it in just one issue, it's really hard to keep track when the same topic is scattered over multiple issues.

Can you provide some details on the versions you are using (deCONZ, iOS, Safari)

From the changelog the 2.05.54 had fixes (tested with iPad Safari 10 and Mac Book Air Safari 11)
 
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_54

In 2.05.59 there were further additions so that the session and login can be restored  based on the url parameter `gwid`, more details here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1238

>why do we need a password just to turn lights on or off via the web GUI. You do not password protect a wall switch!

The simple answer is, because it's a webapp + REST-API. Without any protection every website you visit would be able to control your lights.

>Please either gives us the option to disable password, or better still only password protect changes to the system, like adding, removing or changing resources, just not basic day to day functions!

Sorry disabling password, ain't gonna happen the risks are really high.

I think the main problems needs to be checked, which is login is not remembered.

You may check with a recent version and create a bookmark while logged in with the url of for example the main page (so that it contains gwid parameter), this should already work.

daaaaa, 2019-03-21:
When I try to save the bookmark on my iPhone desktop iOS loses the query string part, consequently the gwid.

I was wondering if [this is related](https://stackoverflow.com/questions/52328733/pwa-loses-its-params-and-query-params-after-being-added-to-the-home-screen).

ademobile, 2019-03-22:
I am using v2.05.60, for Mac (10.14.3) with Safari 12.0.3 and iOS 12.1.4 (i.e. all STANDARD supplied versions). I find the saved book mark still asks for password after a day or two (sometime even next visit).

At the end of the day, a setting to disable password is my risk to take and as I have mentioned before the issues I am suffering with IKEA sensors going to sleep is causing arguments and you should offer people a 'hidden' setting to help them. The likelyhood of issues is remote and if managed correctly is nearly no existent. I cannot stop visitors to my house flicking the light switch on or off, so if the option was given I am in control to enable or disable it as I see fit.

If not I am going to have to end up writing a web front end to allow the direct access I need, however its time I don't have at the moment!

Larsn1, 2019-06-22:
Meanwhile I am on 2.05.66 and still have to type in the password every time I login. Mostly safari as I don’t have any other browser yet

Smanar, 2019-06-22:
Wich one url are you trying ? Have you "gwid" in the url ?

Larsn1, 2019-06-22:
http://192.168.0.xxx/pwa/login.html

closed 2019-10-27 by stale[bot]