dresden-elektronik/deconz-rest-plugin #1369: Unable to login into Phoscon App with default password or reset password

stefanreimers, 2019-03-20:
Finally I found the time to install Raspbee in my Pi3. I used an existing Pi installation running Mozilla IoT Gateway as well.
Unfortunately I can not login into Phoscon UI with the default password mentioned on https://github.com/dresden-elektronik/phoscon-app-beta. Any attempt to reset the password fails with 
"Failed to reset password
Make sure the gateway was powercycled recently"

Fact is I restarted a) the services b) even the PI several times and cannot reset the password anyway. Is there a second way to set a new password via command line? Or else?

manup, 2019-03-20:
If this is a very fresh installation with no prior set password there should be a prompt to set a new password. Looking like this: https://doc.phoscon.de/app/doc.html#commissioning-login-password-en

What browser are you using? You may also check that no script blocker or similar is hindering the sites Javascript from executing.

If you are using the dresden-elektronik.de/pwa site please try the local hosted Phoscon App:

    http://<raspberry ip-address>/pwa

>Is there a second way to set a new password via command line? Or else?

Easiest is to delete the zll.db file (while deCONZ isn't running), but on a fresh install it won't have an effect.

stefanreimers, 2019-03-22:
Thanks for the reply.

Using **another port** solved my issue. 

As mentioned above I am running Mozilla IoT on the same RPi, HTTPS web server included. Although the PWA got delivered by the deCONZ web server, any "password resetting" requests failed.

closed 2019-03-22 by stefanreimers