dresden-elektronik/deconz-rest-plugin #1370: Phoscon und HUE Outdoor Bewegungsmelder

robert-zepnik, 2019-03-20:
Liebe Community!
Ich kann den neuen Philips HUE Outddor Bewegungsmelder nicht in der Phoscon App integrieren!
Bitte um Hilfe!
DANKE!
robert

manup, 2019-03-20:
English please, this is a very  international community :)

The Hue outdoor motion sensor was added in deCONZ version 2.05.59

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Are you using 2.05.59 or 2.05.60?

If the sensor can't be added via Hue motion sensor integration wizard, you can try adding it via *Menu > Sensors > Add new sensor > Others*

robert-zepnik, 2019-03-20:
Sorry, I am NEW here and to Zigbee!
2.05.47...
I will look for the latest version an how to install it!
Thank you!

robert-zepnik, 2019-03-21:
The Phoscon App Updates only to 2.05.54.
😫
Why?

How Can I update manually (beginner!)? 
THX!

manup, 2019-03-21:
Ah yes sorry, the latest 2.05.60 isn't yet in the "simple" update channel.

To update manually follow steps 1–3 from the install instructions:

https://github.com/dresden-elektronik/deconz-rest-plugin#installation-raspberry-pi

**Important** always make a backup first.

robert-zepnik, 2019-03-21:
This is quite complicated, USB connection needed. 🤨
Why can't it be updated via Phoscon App???

robert-zepnik, 2019-03-22:
Sorry, I can't even connect the Phoscon Gateway to my Win10 PC with deCONZ!
I bought a USB A-A cable, but on deCONZ start page I am not able to connect!

robert-zepnik, 2019-03-25:
Pls help! I would like to update my Phoscon Gateway but cannot connect via USB. 
Do I need a "crossed" cable?
What can I try else?
Is there another way to connect to deCONZ in a Win10 Notebook?
THX!

ebaauw, 2019-03-25:
What hardware do you have?  With a RaspBee, you can only run deCONZ on a Raspberry Pi.  To run deCONZ on anything else, like Windows, you need a ConBee.  Both have the same Zigbee hardware and firmware.  The ConBee comes as a USB stick, the RaspBee as a Raspberry Pi shield.

You can only “connect” to deCONZ through the REST API, there’s no remote USB connection.  The GUI only runs on the local machine, but you can use screen sharing (VNC) to access it remotely.

robert-zepnik, 2019-03-25:
I have a Phoscon gateway which should be a Raspberry Pi!? (The "Help-doc" of deCONZ for Windows describes connecting via USB...)
Anyway: I should connect the gateway to a Monitor via HDMI, Keyboard&Mouse via USB and work directly on the gateway? Is there a special instruction?
Thank you! 
rz

ebaauw, 2019-03-25:
That would work.

manup, 2019-03-25:
@robert-zepnik Quick note: The 2.05.60 update will be available via Phoscon App update till weekend.

robert-zepnik, 2019-03-26:
Thank you!

2019, 23:46:

robert-zepnik, 2019-04-04:
Thanks for updating the Phoscon App!
I already connected the Philips HUE Outddor Sensor.
I already added a movement detection command:
Movement = 100%
After 1 Minute w/o movement = 50%
After 2 Minutes = 5%

The settings for the sensor: active from dawn til dusk

Unfortunately I cannot add a daylight command (create button is grey and mouseover sign 'not allowed')!!!

Lights should be off over the day, 5% on from sunset til sunrise and dimm to 100% on detection of movement (see above).

Can anyone help please!
Thx!
r.

closed 2019-08-09 by stale[bot]