dresden-elektronik/deconz-rest-plugin #1373: Retrieve id's for use with openhab

posixx, 2019-03-22:
Hello,

As the discovery of devices is broken in the current binding implementation in openhab, i want to add devices manually. But, in order to be able to do this, we need the id for every device. As the backup files are binary; how cna we retrieve the id's of devices / groups within deconz?

As an example of manually add them to openhab:

Bridge deconz:deconz:homeserver [ host="192.168.0.10", apikey="ABCDEFGHIJ" ] {
    presencesensor      livingroom-presence     "Livingroom Presence"       [ id="1" ]
    temperaturesensor   livingroom-temperature  "Livingroom Temperature"    [ id="2" ]
    humiditysensor      livingroom-humidity     "Livingroom Humidity"       [ id="3" ]
    pressuresensor      livingroom-pressure     "Livingroom Pressure"       [ id="4" ]
    openclosesensor     livingroom-window       "Livingroom Window"         [ id="5" ]
    switch              livingroom-hue-tap      "Livingroom Hue Tap"        [ id="6" ]
    waterleakagesensor  basement-water-leakage  "Basement Water Leakage"    [ id="7" ]
}

So we need to get the id somehow..

ebaauw, 2019-03-22:
A GET of `/lights`, `/groups`, or `/sensors` returns all lights, groups, or sensors, including the id.

posixx, 2019-03-22:
duhh. that simple.. tnx Erik!

posixx, 2019-03-22:
Oh, my Hue TAB's are not within the sensors output. /devices does return empty brackets. Any idea?

ebaauw, 2019-03-22:
/devices is still under development.  The Hue tap should be in /sensors.  Maybe you need to re-pair it?

posixx, 2019-03-22:
nope they are not. but i found the ID's using homeseer where i'm migrating away from.

closed 2019-07-27 by stale[bot]