dresden-elektronik/deconz-rest-plugin #1374: Gledopto Lights Not Meshing

bdb12, 2019-03-23:
Hey Guys

So I have mostly Gledopto lights. The models I have are GL-S-003Z & GL-B-008Z both version 1.0.6 in Phoscon. 

The issue I have is that some that are further away will not connect or loose connections and even the lights that stay connected do not mesh. This is s problem as I have a large house and the Conbee is in the center of the house. I thought the whole idea is that Zigbee is supposed to mesh. The other issue is that I need my other non routing sensor devices to connect to the Gledopto lights. My Smartthings hub seems to connect to these same lights and it's in the same position.  see the picture. Unlike others I have seen all my devices only connect direct to the conbee and not to each other.
[image]

BTW the conbee is running on a windows PC running Windows 10, It connects to Home Assistant. I am running the very latest versions on both. Version 2.05.60 / 3/10/2019
Firmware 26320500

One other thing. If coverage is an issue should I have multiple conbees set up as routers? Would they have to be plugged into a a Pi or can they simply be plugged into a USB power supply?

Please help! Thanks in advance!!

ebaauw, 2019-03-23:
Some thoughts:
- Did you double-check that the check marks are set under CRE in the deCONZ GUI?  Do the numbers appear next to the lines when you enable LQI?
- Are you using an USB extention cable for the ConBee?  We’ve seen radio interference when the ConBee is plugged in directly.
- Do you keep your lights powered, or are you using traditional wall switches?
- Is the Smartthings hub on the same network?  Having multiple Zigbee gateways on the same network might cause issues with devices that only report to a single destination.

bdb12, 2019-03-23:
Thanks for the response

Did you double-check that the check marks are set under CRE in the deCONZ GUI? Do the numbers appear next to the lines when you enable LQI?

A: Yes to both. What do the numbers mean?

Are you using an USB extention cable for the ConBee? We’ve seen radio interference when the ConBee is plugged in directly.

A: I am using a USB extension, though it is only 200mm long
Do you keep your lights powered, or are you using traditional wall switches?

A: Yes I have traditional switches but they are on. Not 24/7 but certainly while I am testing. 

Is the Smartthings hub on the same network? Having multiple Zigbee gateways on the same network might cause issues with devices that only report to a single destination.

A: Yes the Smarthings is on the same network but at one point i turned it off for a few hours as I thought that might be a reason.

ebaauw, 2019-03-23:
>  The models I have are GL-S-003Z & GL-B-008Z

These are bulbs/spots aren't they?  Did you install them into metal enclosures?  That could interfere with the radio traffic (courtesy of Faraday).  Could you move two of the bulbs close to each other (couple of meters, in the same room, straight line of sight) and see if deCONZ shows a connection between them?  This might take some time - give it 30min or so.

I have no experience with these models, but I'm told they have the same Zigbee hardware and firmware as the RGB+CTT controllers (GL-C-008).  I only have one test unit, but it meshes alright with other lights.  The single GLEDOPTO floodlight (with RGBW hard- and firmware) in my production network also meshes just fine with other lights.

> What do the numbers mean?

The Link Quality Indicator, from 1 (very bad) to 255 (very good).  I don't know exactly what it means physically, but it's a measure for the quality of the radio connection from one device to another and the other way round.  If the numbers appear, deCONZ is receiving some routing info.  Do the numbers appear on both sides of the `/` (i.e. does deCONZ receive info from the ConBee as well as from the GLEDOPTO)?

> though it is only 200mm long

That should probably be enough to deal with any interference from the PC's Wifi, bluetooth or internal electronics.

> Not 24/7 

You want your Zigbee routers to be powered 24/7.

> Yes the Smarthings is on the same network but at one point i turned it off for a few hours as I thought that might be a reason.

I'm probably being paranoid, but you might need to reset and re-pair the GLEDOPTOs to deCONZ after turning off the SmartThings hub, to undo any configuration by the SmartThings hub.

bdb12, 2019-03-23:
Thanks again for the explanation. Yes they are all bulbs and have nothing round them except a fabric lamp shade. I will try another longer USB extension. I have tried to many time reset/remove/re-add the lights from both Phoscon and deconz. 

Is there something else I can do to boost the Zigbee signal?

bdb12, 2019-03-23:
PS They (the lights) were never connected to Smartthings.

ebaauw, 2019-03-23:
> Is there something else I can do to boost the Zigbee signal?

Keep it clear from others using the 2.4GHz band (WiFi, bluetooth, DECT, your microwave); try a different channel (25 has least overlap with WiFi bands).

Move the lights closer; or place additional lights (or other Zigbee routers) in between the ConBee and the lights.

bdb12, 2019-03-24:
So I have turned off smarthings, turned off my Dect phones. Restarted the PC that runs Deconz, tried different USB ports. I have also just tried it on a completely different machine. It appears to have got worse and I would suggest I might have a faulty unit as I have lost all links to everything despite them some items being within 30 CM from the Conbee stick. 

Each time I have made a change I have done 1 thing at a time. Is there a way to re-flash the whole stick? I am running it on a windows 10 machine.

bdb12, 2019-03-24:
PS really appreciate the quick responses BTW :)

bdb12, 2019-03-24:
see latest pic 
[image]

closed 2019-07-29 by stale[bot]