dresden-elektronik/deconz-rest-plugin #1375: Questions about my mesh (unknown devices/bad route)

pergolafabio, 2019-03-23:
ok, in my screenshot below, you see i have a verry easy mesh:
https://www.dropbox.com/s/rulw8bmpyqek4rz/meshxiaomi2.JPG?dl=0
deconz conbee FW: 262F0500

so , in phoscon i added:
3 x osram powerplug, i use them as a repeater to create better mesh, so i have them on each floor
2 x ikea remote switch
4 x xiaomi pir body sensor

the red lines are my floors
so,
ground floor (1 x xiaomi , 1 x osram)
first floor (2 x ikea , 3 x xiaomi , 1 x osram)
second floor (conbee + 1 x osram)

2 issues/questions

1) my xiaomi? 3 of them are paired directly with the conbee, while the osram power plug1 is verry nearby... xiamo_sensor_4 , well thats is giving me headaches, its verry close to plug1, why is it choosing the worst route ever? restarting conbee doesnt help? how can i relink it? 
also took out the battery, doesnt help...  (all new battery)
sensor4 isnt working reliable!

2) those 3 devices in the red circle, i have no clue what they are? dont have them in phoscon, what are they doing there? there is even one linked
i had 3 x ikea pir sensors before, but they sucked, so remove them from phoscon and took out batterys, maybe they are remaing stuff? but then again, why is there still 1 linked?

thnx in advance

Smanar, 2019-03-28:
Lol, I have same problem, I have a realy bad router, and all devices want to use it as router.
The only way I know, is moving the device close the device I ant to use as router, and launch the pairing procedure. If it fail, I need to delete the device (phoscon and deconz) and restart. Or I power off the bad router before the pairing procedure.

IDK for ikea, but on xiaomi device, the mac adress is written on the device, you can compare it with theses one you have in your mesh.

If you want to delete a device sometime (I think it depend on version and device) you need to delete it in phoscon AND deconz.

pergolafabio, 2019-03-28:
@Smanar  ok, thnx for feedback

about the mac address, ok gonna look it up
but those ikea are deleted from phoscon webapp, in decozn, you can also delete the node, didnt know
so i deleted those 2 x unlined nodes, after 1 day there were  back??? wtf?

and that 3th node with 0xd860 , i have no idea that is, also if delete that node in deconz , it just comes back AND linked? go figure

also question: do we have actually control about the route that an end device is using? i linked all my end devices with the conbee stick itlself, after it i just placed them in correct rooms

Smanar, 2019-03-28:
I have similar problem with a xiaomi switch (even with battery removed) but it go out after some days or a deconz reboot (don't remember).
All 3 nodes are back if you delete them with the del key ? If you select them you don't have more info in node panel ?

Nope, no control on route, I have ask for exactly the same question ^^. Its coded in firmware, the only method I know is to put the device close the router I want it use.

manup, 2019-03-28:
>my xiaomi? 3 of them are paired directly with the conbee, while the osram power plug1 is verry nearby... xiamo_sensor_4 , well thats is giving me headaches, its verry close to plug1, why is it choosing the worst route ever? restarting conbee doesnt help? how can i relink it?
also took out the battery, doesnt help... (all new battery)
sensor4 isnt working reliable!

Once Xiaomi battery powered devices have picked a parent, they will stubbornly use it.
The best approach is to just start sensor search and reset the Xiaomi device at it's actual location.

Important: You **don't** need to delete the device first, just repair it so it picks another parent.

pergolafabio, 2019-03-28:
@manup  @Smanar but i see in my deconz mesh, that sometimes my battery end device is linked with another router, so why is repair/relink needed with a router? it changes anyway?

see this screenshot
there is no node info, i have no idea what the device is, if i delete the node, after 1 sec its back and linked
https://www.dropbox.com/s/g90mnwrtbpnxz2j/nodemesh.JPG?dl=0

i dont see it it in phoson  either? is it maybe that virtual daylight sensor? thats the first time created?
is there no website where you can do some kind of mac lookup like an ordinary mac address of an ethernet adapter ?

Smanar, 2019-03-29:
0x90fd9.... are ikea bulb on my side
Virtual daylight don't appear

> but i see in my deconz mesh, that sometimes my battery end device is linked with another router, so why is repair/relink needed with a router? it changes anyway?

So you are saying the xiaomi sensor change router itself ?

pergolafabio, 2019-03-29:
i only have 2 x ikea remotes, thats all , they start with 0x000d6
0x90fd9 , still no clue about it, there is no name / model ; dont see it in phoscon etiher, if i delete the node, its instant back and also linked back
i have no clue

and yes, my end devices like ikea remote switch /  and xiaomi sensors, they change routers automaticly, i think that its normal.... normally the first time , i always though, you need to pair it with conbee gateway, afterwards, it will find the nearest router anyway, so i was told, and it also seems to work that way...

Smanar, 2019-03-29:
@manup, I already have same situation too, but I thought I had seen badly.
1 Temperature sensor and 1 door sensor (Xiaomi) meshed trought an ikea bulb, have leaved it for a direct connexion to conbee after somes hours. So either the mesh is not reliable the first hours after pairing, either the xiaomi device can change router during the first hours.

Smanar, 2019-03-29:
I think the sensor is too much incomplete for that, but have you tried to see it wih the API ?

pergolafabio, 2019-03-29:
what do you mean? see it with API?
here is screenshot about node info : https://www.dropbox.com/s/ymt0c42swmwrnof/nodeinfo2.JPG?dl=0

Smanar, 2019-03-29:
Lol, realy incomplete device. Don't loose time with the api, I think it will be useless.
On my side, I have delete it somes days after, but I don't remember what I have do.
Have you tried to unplug the plug3 ?

pergolafabio, 2019-03-29:
no, not tried yet :)

will it make a a difference? why do you want me to test that ?

Smanar, 2019-03-29:
Random test ^^.
On your mesh your "ghost ressource" go trough the plug 3, so perhaps a "router reset" can kill your ghost.

A more technical explanation (but take care I m amator).
The ghost device can be just some data in a "neighbour table", but I can say if data are in gateway or router. It will disapear itself with time

pergolafabio, 2019-03-29:
ok, thnx for info :;)

gonna remove plug3 for a day, see what happens if i put it back
reset conbee gateway? is that a factory to defaults? or just a quick reset, dont wanna loose my stuff :)

Smanar, 2019-03-29:
Nope, you don't need a day. Just plug/unplug to test
It will take some days if you dont touch anything.
And no, don't reset the gateway, just turn off/on.

But like I had say before, you can just let all in state and wait for some days.

pergolafabio, 2019-03-29:
ok, will do some testing, thnx for the information :)

pergolafabio, 2019-03-30:
no mather what i do, unplug all plugs, unplug the conbee,reboot
those 3 devices are always back, and one is always linked :)

Smanar, 2019-03-31:
I don't know how are memorised this kind of data, "ephemeral" memory or not.
Just wait for some days, it have worked on my side. If in some days your ghost devices are still here, you will need someone with more knowledge than me.

closed 2019-08-05 by stale[bot]