dresden-elektronik/deconz-rest-plugin #1376: Cant edit aqara cube in switch editor

TangoTaker, 2019-03-23:
Hi, im not sure if I missed something but how can i edit the Aqara Smart Cube in Phoscon, I can edit every Switch in the switch editor except for the cube.
Thanks

closed 2019-03-23 by TangoTaker

manup, 2019-03-24:
The Xiaomi Cube isn't yet integrated in Phoscon App switch editor. It is on the todo list and there is already a concept but no ETA yet.

froggy974, 2019-11-20:
i see that cube editor have been added in V2_05_70 but it's still missing from switch editor.
Is the integration not fully done yet ?

manup, 2019-11-20:
The cube is supported in the switch editor. You may need to repair it though, so that all resources are created. The recommend way is to delete it under  *Menu > Devices > Switches* and then repair it.

froggy974, 2019-11-20:
thanks for the feedback,
i deleted it, and didn't find any "repair" button
so i'm trying to add it again , via phoscon, switch, add new switch, other ... i keep the link buton pressed until bliknig on the cube, but nothing happen in phoscon web app.

did i missed something ?

TangoTaker, 2019-11-20:
Hold the button for 5 seconds until it flashes. Then you should shake and rotate it in order to keep it active. 
I can find the cube in my switch editor, but I cant add any action to it, Phoscon always tells me something went wrong. Does anybody else experience this?
Regards