dresden-elektronik/deconz-rest-plugin #1377: Offset for Aqara Temp/Hum/Press Sensor

kugelkopf123, 2019-03-24:
I've found a solution to change the offset. I tried an offset of -10 but nothing changed on the Temp value.

command:`curl -H 'Content-Type: application/json' -X PUT -d '{"offset": -10}' http://192.168.1.73:8080/api/ABCDEFG/sensors/2/config
`
Result:` {"config":{"battery":100,"offset":-10,"on":true,"reachable":true},"ep":1,"etag":"469a3fa20640dca88f0240552a054701","manufacturername":"LUMI","modelid":"lumi.weather","name":"Sensor2","state":{"lastupdated":"2019-03-23T12:56:46","temperature":2302},"swversion":"20161129","type":"ZHATemperature","uniqueid":"aa:15:55:00:x2:2c:cy:a3-01-0402"}`
[image]
[image]
[image]

ebaauw, 2019-03-24:
`config.offset` is in 0.01°C, just like `state.temperature`

kugelkopf123, 2019-03-24:
> `config.offset` is in 0.01°C, just like `state.temperature`

Thank you!! It works! But is there a special reason that the range from the offset ist +/-500 ? eg. +/- 5.00°C

ebaauw, 2019-03-24:
No special reason.  It seemed like a reasonable range to calibrate the sensor.

kugelkopf123, 2019-03-25:
May I ask if it is possible to make this value freely selectable? Especially in the field of Humidity 5% are not the most. As far as I've seen, there is no offset at Pressure ?!

closed 2019-07-30 by stale[bot]