dresden-elektronik/deconz-rest-plugin #1378: Xiaomi MCCGQ01LM Door Sensor and DJT11LM Vibration Sensor Become Unreachable After Inactive For Long Duration

jerrychong25, 2019-03-25:
Hi,

I would like to report an issue for **Xiaomi MCCGQ01LM Door Sensor** and **DJT11LM Vibration Sensor** become unavailable in **Home Assistant** and **deCONZ Phoscon App** after inactive for long duration, as per screenshot below:
- deCONZ Phoscon App Screenshot
[image]
- Home Assistant Screenshot
[image]

MCCGQ01LM Door Sensor Activity Log in Home Assistant:
_25 March 2019 4:30am Unreachable (Raspberry Pi Restart This Time)
25 March 2019 5:22am Reachable
25 March 2019 9:34am Unreachable (Raspberry Pi Running As Usual)
25 March 2019 9:48am Reachable_

DJT11LM Vibration Sensor Activity Log in Home Assistant:
_25 March 2019 4:30am Unreachable (Raspberry Pi Restart This Time)
25 March 2019 4:58am Reachable
25 March 2019 9:36am Unreachable (Raspberry Pi Running As Usual)
25 March 2019 9:45am Reachable_

Could advise us how to solve this issue? Or this is a normal behaviour for deCONZ REST plugin?

Thanks.

heitec11, 2019-03-26:
We have the some problem.
See [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1354](url)

manup, 2019-03-26:
Inactivity shouldn't be a problem, the sensors check-in once per hour. For example one of  my window sensors was wasn't triggered in months but still is active.

The no.1 problem for Xiaomi sensors is when their parent routers (e.g. lights) are either powered off, rejoined with a new address, or otherwise lost the children. The Xiaomi sensors most often do not search a new parent and try to route stubbornly through the old one.

The simplest approach to fix such a device is to start sensor search and reset the sensor (**don't** need to delete the sensor).

jerrychong25, 2019-03-26:
> Inactivity shouldn't be a problem, the sensors check-in once per hour. For example one of my window sensors was wasn't triggered in months but still is active.
>
> The no.1 problem for Xiaomi sensors [...]

Yup, once the sensor have events detected, it will active back. 

But problem is that Home Assistant and deCONZ Phoscon App does not reflect latest states, which will make user confused.

Could advise how to start sensor search and reset the sensor in your case? Is this done via API call or Home Assistant settings?

Thanks.

chimpy, 2019-03-26:
Sensor search is done in the Phoscon UI. Resetting of the sensor is done via the little hole on the device itself.

jerrychong25, 2019-03-27:
@chimpy,

Thanks.

Hope there is another solution for fixing this automatically instead of manually search and reset the sensor every time.

As we can't predict when these sensors will become inactive.

jerrychong25, 2019-06-24:
Since this issue does not affect my Raspberry Pi performance, therefore I close this issue.

closed 2019-06-24 by jerrychong25

xbmcnut, 2019-08-06:
This is still an issue for me under 2.05.66 and Conbee II. Can the issue be re-opened?
[image]