dresden-elektronik/deconz-rest-plugin #1379: Maximum Number of Sensors Can Be Added Into deCONZ Phoscon App And Home Assistant?

jerrychong25, 2019-03-25:
Hi,

I would like to add large number of sensors (> 20 sensors) into deCONZ Phoscon App.

Could I know what is the maximum number of sensors can be added into deCONZ Phoscon App by default?

At the same time, what is the maximum number of sensors can be imported from deCONZ Phoscon App into Home Assistant by default?

Thanks.

Kane610, 2019-03-26:
There is no limitation of devices in the Home Assistant integration. The rule of thumb for deCONZ is that it supports around 200 devices, how many which are routers/sensors out of that I don't know.

ebaauw, 2019-03-26:
There is a limit to the number of end devices (typically battery-powered) that can be connected to a single router (typically mains-powered).  This limit is set in the router's firmware.  They bumped it for the RaspBee/ConBee (I think to 16 or 32, but I'm nor sure).  If you want to add more end devices (sensor), you need to add more routers (lights) first.

jerrychong25, 2019-03-26:
Hi @Kane610 and @ebaauw,

Thanks for the info.

closed 2019-03-26 by jerrychong25

vandalon, 2019-09-04:
Out of curiosity, is the end-device-limit still 34? I'm reaching that point :)

SuperSchelli, 2019-09-28:
my rest api reached 99 devices when i query the sensors. i tried to add a new switch, but it kicked another out of it's place... I have some empty spots in the Rest api, meaning those numbers are not used at all. I have loads of light bulbs in there, more than 30, so it shouldnt be a problem to have loads of battery power devices i guess... has anyone ever had a number >99 for a rest api call ?

ebaauw, 2019-09-28:
I hacked the database to enforce a numbering scheme on the device IDs (floor/room/device).  No problem to use device IDs above 100.

thomasvnl, 2022-06-27:
> my rest api reached 99 devices when i query the sensors. i tried to add a new switch, but it kicked another out of it's place... I have some empty spots in the Rest api, meaning those numbers are not used at all. I have loads of light bulbs in there, more than 30, so it shouldnt be a problem to have loads of battery power devices i guess... has anyone ever had a number >99 for a rest api call ? meaning the id of the device being 100 or greater ?

Experiencing the same issue. Cannot add new sensors because id = 99. Do not have 99 sensors added as in the mean time some sensors have been removed/re-added but still got new ID's (no clean up in isle deCONZ?). Asking myself if I want to go through the hassle of added all my devices, once more, or just buying a Sonoff Zigbee stick and just using Zigbee2MQTT from here on out...

Mimiix, 2022-06-28:
Sensors do not have a limit. 

You might have too many end nodes opposed to routers. 

Feel free to open an topic on the forums to go into more depth and see why it works.

thomasvnl, 2022-06-28:
> Sensors do not have a limit.
> 
> You might have too many end nodes opposed to routers.
> 
> Feel free to open an topic on the forums to go into more depth and see why it doesnt work.

Thanks for the pointer. I have three routers that I've tried to evenly distribute throughout my home, but I might need to check if they are all functioning properly/routing devices in a balanced manner.

Mimiix, 2022-06-28:
There's your issue. Routers can have between 10 to 20  nodes. Coordinator around 25.

You really need to add more routers :)

manup, 2022-06-28:
> Sensors do not have a limit.

There is actually a limit on the ID but it is 10000, should be enough :)

I also think there likely need to be more routers in the network.