dresden-elektronik/deconz-rest-plugin #1380: More button events in Phoscon Switch editor for Aqara WXKG03LM Rev.2 and WXKG02LM Rev.2

bohtho, 2019-03-25:
I noticed the Phoscon app at http://www.dresden-elektronik.de/pwa/ now supports showing these Aqara switches as connected sensors, *and* makes them available in the Switch Editor. Great stuff!

However only with simple button events (no double press or hold), which makes the actions available to eg. Trådfri Remote Controls, like "hold for dim up/dn, press for on/off toggle", *sadly* *not* available for the Aqaras yet (which would have been perfect because of their "living room look"!). 

Switches: Aqara WXKG03LM Rev.2, and WXKG02LM Rev.2

manup, 2019-03-25:
One problem here is that these switch models don't support a button release event. So implementing long press dimming isn't possible, since we don't receive a stop dim command.

The double press events on the other side will be supported in future releases, they are already on the todo list.

bohtho, 2019-03-25:
That’s true. The long press would still be a useful event along single and double press if one could set it to dim eg. 10% or a fixed amount on long press (or activate a scene or anything).

In a perfect world a long press would initiate a slow dim in the opposite direction of last time which would stop the dimming at a double press (or a single press if Deconz could differentiate between single presses while dimming and otherwise). A single press could then be usual power toggle.

chimpy, 2019-03-26:
> In a perfect world a long press would initiate a slow dim in the opposite direction of last time which would stop the dimming at a double press (or a single press if Deconz could differentiate between single presses while dimming and otherwise). A single press could then be usual power toggle.

This is exactly what I achieve with the round button switch using deconz/HA/Node Red.

bohtho, 2019-03-26:
Do you mean “single press release” when you say “stopping on release”? Have you just programmed a trigger in node-Red to continue dimming in increments by itself until the next single press release is received? And a variable saves the direction to inverse it next time? I currently just dim a fixed amount one time in opposite direction on hold. Doesn’t give me much dimming range though. And I don’t feel the performance of Node-Red and the REST API is as quick as I like, or as a resource link and rules directly in Deconz would be.

chimpy, 2019-03-27:
Single press release is used for toggling (because single press always triggers regardless of long or short press).
Dim starts at long hold and stops at long hold release:
[image]

Schema of the dimming logic in a subflow node:
[image]
Code: https://pastebin.com/raw/vTSfhsZS

Fibonacci is used because I figured the longer we press, the more the dimming should accelerate to get where we want. It offers control over a short period whilst not taking aaaages to get to max brightness from a dim state.

mink2k, 2019-04-01:
is there a way to activate the double press event by my self to show on the Phoscon app or are there any beta programms we can join to activate this feature?

capstan1, 2019-06-20:
I'm using HA and NodeRed as workaround right now. You could try that.
But still keeping fingers crossed for support of long press and double press in Phoscon.

closed 2019-10-25 by stale[bot]

MarcProe, 2019-11-18:
> The double press events on the other side will be supported in future releases, they are already on the todo list.

Any news on this? Those switches are really great for the price, implementing the other events in Phoscon would be greatly appreciated.

SamB-GB, 2019-12-17:
> > In a perfect world a long press would initiate a slow dim in the opposite direction of last time which would stop the dimming at a double press (or a single press if Deconz could differentiate between single presses while dimming and otherwise). A single press could then be usual power toggle.
> 
> This is exactly what I achieve with the round button switch using deconz/HA/Node Red. Dimming lasts as long as the button is pressed (accelerating in Fibonacci increments), stopping on release. Single click is still toggle.

But doesn't the round button issue a release button scene?

chimpy, 2019-12-20:
I'm not sure what you mean. When the button is released (long click release), the dimming stops.

SamB-GB, 2019-12-20:
The models in the subject don’t send a long click release

BrownMuffin, 2020-02-20:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG02LM-2018

These switches send different events like double click and long press, can't these be catched?

MichaelKorn, 2020-03-14:
I have WXKG02LM switches and all 9 events are working with ioBroker. But Phoscon just shows the 3 "single press" events.

MarcProe, 2020-05-24:
@manup looks like the new Oppo Switches send a [release](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1380#issuecomment-476401708), maybe this can be revisited?

capstan1, 2020-07-31:
Any news on the single and double rocker switches? Having all these double and long presses managed in Node Red really slows the system down. Would be way better to have everything inside Phoscon/Deconz

jozemlakar, 2020-11-04:
Looking at this as well. I am a dev, so if someone points me in the right direction, I might move myself and do something about it.