dresden-elektronik/deconz-rest-plugin #1384: Phoscon - specific time for motion sensor

maikkoe, 2019-03-27:
Hi all,

I configured a Tradfri Motion Sensor for one of my groups. It works correctly, but i want to use a specific time for eg. from 8pm till 11pm and not 'all time' or 'from dusk till dawn'. Are there any possibilities?

chimpy, 2019-03-27:
Use Home Assistant.

realwax, 2019-03-29:
Thanks for your comment. Yes the requests are very similar. Mine requests in addition to set a scene as action not only a dim level or on off state by motion sensor. This would really value add phoscon add as one gateway to rule them all ;) in terms of functions in comparison to other bridges and apps. it is clear that complex flow should be done ourselves in iobroker,fhem, and others. Select timeframe and scene on motion sensor should be not to hard isnt it?

lnoc, 2019-04-22:
> Hi all,
> 
> I configured a Tradfri Motion Sensor for one of my groups. It works correctly, but i want to use a specific time for eg. from 8pm till 11pm and not 'all time' or 'from dusk till dawn'. Are there any possibilities?

I would like this feature as well. A time range option would be great!

realwax, 2019-04-22:
Should not be too tough for devs to add this feature in a future release. Currently it's not available. You can help yourself with e.g. iobroker, deconz plugin, javascript plugin and a blockly/javascript solution. I can share mine if desired.

realwax, 2020-02-05:
rebump

Kugelfang666, 2020-03-10:
Currently I have this functionality with my hue bridges using iConnectHue. Not having the ability to define timeslots is actually what keeps me from fully migrating to deconz