dresden-elektronik/deconz-rest-plugin #1386: Sensors - Limit of 16 in Phoscon APP ?

holdestmade, 2019-03-27:
Is there a limit to the number of sensors in Phoscon APP ? I have 16 various Aqara Sensors, if I add another, it replaces the last one in the list.

chimpy, 2019-03-27:
This is normal. You must add a router device in order to continue adding child devices.

holdestmade, 2019-03-27:
Can you give me an example please ? I have many lightbulbs/ sockets and switches already in the Phoscon APP.

netzbus, 2019-03-27:
I am also interested in this. I thought there is no "limit"

holdestmade, 2019-03-27:
I have since looked and the WIKI states max 32 devices. I have 42 but a mix of mains and battery devices

manup, 2019-03-27:
The limit described in the wiki is for direct connected end-devices, though I need to check with latest firmware – when I think it was lowered to 20 or so since some memory limits were hit.

Anyway, the limit can be easily expanded by adding more routers like lights and power plugs to the network. One of our test networks has ~180 devices. There are also quite some users with networks containing more than 120 devices.

The hardest part are reliable routers, in my experience the most stable devices are Philips Hue lights.

holdestmade, 2019-03-28:
I have just ordered a Zigbee USB that apparently can have router firmware loaded and that used, cheap from Aliexpress ~£4

I'll post if it works ok

Smanar, 2019-08-19:
@manup just by curiosty, do you know the actual limit for maximum direct-connection ? And if this number is the same for the gateway and routers ?