dresden-elektronik/deconz-rest-plugin #1387: Loosing all xiaomi device after a deconz power off.

Smanar, 2019-03-27:
Hello,
so, first all is repared ATM, but I prefer to know how to avoid this situation.
I have just power off deconz for 24h, and when I power it to on, I have loose all xiaomi sensors/switches. press button 1 time don't work, 5 press don't work, key '0' not work too.
The only way I have found to repair is repairing all devices one by one.

It's something normal? I have do something bad ?

chimpy, 2019-03-28:
Read other issues. Xiaomi devices do not change parent node easily. So if you have other router devices on your ZB network, and deconz assigns the Xiaomi child nodes to them upon restart, the Xiaomi devices will still try to route through their old parent nodes, to no avail.
This lasts until you re-pair them.

Smanar, 2019-03-28:
Yep, but I have loose all my Xiaomi devices, theses one that uses routers and theses ones that have direct connexion to conbee.

I haven't the xiaomi gateway to check, but for me it's not normal, it mean if you turn off the gateway for 24h, you need to repair all your devices just after ?

You already have tried to shut off your gateway for 24 h ?

manup, 2019-03-28:
Hard to tell what happened, I guess this can only be investigated by recreating the situation and let the sniffer run the whole time.

Perhaps the Xiaomi devices leave the network or go in a safe state after the gateway can't be reached for some time?

Smanar, 2019-03-29:
> Perhaps the Xiaomi devices leave the network or go in a safe state after the gateway can't be reached for some time?

I do not hope that, because I don't wanna spend my time to remake all the installation if I need to power off the gateway. "Safe state" is possible but I haven't find a way to retreive them.
But I can't understand, I m the first one who stop the gateway for 24 h ? Or I m the first one with this kind of problem ? This issue have 2 days and no one seem to already have this problem.
I just want to know if it's normal or not, all is repair ATM.

For the moment I m not realy hurried to remake the situation ^^, I have lot of problems since this bug (some sensor have 5/10s latency for exemple), but the next time I do that I would think to put the sniffer.

zoic21, 2019-04-03:
Your not alone with this issue (xiaomi and deconz Key, nous router sévices), on a test système i have same problem but i don't have Time box to investigate... Maybe next week

Smanar, 2019-06-17:
Ok, so I have find someone else with the same problem, electric outage > loose some devices.
Someone know if it happen too on Xiaomi gateway ?

For me it's something normal, if the device loose his parent (electric outage, connexion problem, coordinator freeze, bad planet alignement) after some hours, it leave the network.
But it mean every time I have a power off, I need to repair all devices ? No way to avoid that ?

closed 2019-10-22 by stale[bot]