dresden-elektronik/deconz-rest-plugin #1388: Why deconz is spaming attribute request

Smanar, 2019-03-27:
Hello, on deconz I have a device with the blue circle always flashing, so I have sniffed the traffic, and I have found something strange.

[image]

The device is working perfectly except its a realy bad router, it's a legrand switch without neutral.

Can I do something to avoid that ?

manup, 2019-03-27:
Thats expected. deCONZ polls mains-powered devices state as fast as it can. The only exceptions are devices which support reporting, like IKEA and OSRAM lights (and reporting is enabled for these device within deCONZ code).

It shouldn't cause any harm, the Hue bridge for example does the same, since Hue lights don't support reporting at all.

Smanar, 2019-03-28:
Ha, I haven't check reporting for this device yet.
If philips do that, I m ok to conceive it's a good method (as they have the better system for me), but seriously, if you have 30/40 philip bulbs in a meshed installation, the "zigbee jabber" will be terrible !

BTW thx for information.

Smanar, 2019-03-28:
It seem this device support attribute reporting.
I can read/write config with min/max report interval But how I can enable it and desactivate pooling (if possible)

manup, 2019-03-28:
>If philips do that, I m ok to conceive it's a good method (as they have the better system for me),

Reporting is a better method in my opinion, especially on scale.

>if you have 30/40 philip bulbs in a meshed installation, the "zigbee jabber" will be terrible !

It's actually the same, since only one device is polled at the time. Only the whole roundtrip takes longer.  The alternative would be slow updates on state changes.

It may look like a lot of traffic to a human but it really isn't that much. For example you might spin up Wireshark and just sniff your local WiFi/LAN traffic which is magnitudes higher even if you're not doing anything. Machines are chatty.

>It seem this device support attribute reporting.
I can read/write config with min/max report interval But how I can enable it and desactivate pooling (if possible ? )

>On level control and on/off I have min 0, max 600 by defaut.

We need to enable it in the source code, for that we need more information about the device as described in the wiki: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Smanar, 2019-03-29:
"Machines are chatty."
Yep, I m agree, but for me it's not same, the second one is a complete systeme, the first one is just some bulbs in an abandonned house during the day ^^. For me, just for home automation technologie, it's realy enourmous.

The device is this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883
It's in deconz source code, or REST-api code ?

There is a test I can do to be sure it's possible ? before spending time for nothing ?

Smanar, 2019-04-10:
Just a question, to add the report attribute mode on this device, you need to edit the REST-API code or deCONZ code ?

ebaauw, 2019-04-10:
You could configure it (once) in the deCONZ GUI, not needing any code change at all.  To do this automatically, the REST API plugin code needs to be changed.

Smanar, 2019-04-10:
Yep but from I have understand, this device works ATM like a philips bulb (with spaming command) , and it can work like an ikea one (with attribute report).
I have see how to enable/disable report, but not how to disable the spamming mode.

It will ne nice if it will be possible with deconz, not usefull to do it automatically, the device works fine as it, was just by curiosity.

ebaauw, 2019-04-10:
deCONZ (or rather the REST API plugin) will stop polling the light if it receives attribute reports from it.

Smanar, 2019-04-10:
Ha ?
Nice, so I wil try it tommorow, thx for info.
I realy like this github, every day, something new to learn ^^. With the few documentation there is on deconz, this github is realy essential.

Smanar, 2019-04-11:
Yeah, fantastic, all have worked.
I will make some tests with the device, and I will update the first issue (even I don't see another people with this device on deconz, I realy don't see why, Legrand is realy good manufacture)

But A last question, what are good value for reporting ?
For on/off : I have set 1 - 600 (0 still spamming and 600 is the defaut value)
For level control : I have set 0 - 600 - 0 ( I have just add the 0 value for Reportable change, 1 spamming the websosket return)

ebaauw, 2019-04-11:
I typically use 1, 300 for _On/Off_ (report at most once per second when _OnOff_ changes and periodically every 5 minutes); and 1, 300, 1 for _Current Level_ (report at most once per second when _Current Level_ changes by 1 or more, and periodically every 5 minutes).

The polling mechanism checks how recent the last report was, and based on that determines weather there's a need to poll the light.  Make sure to configure bindings for both the _On/Off_ and _Level Control_ clusters and reporting for both the  _OnOff_ and _Current Level_ attributes.

Smanar, 2019-04-12:
Thx a lot for tutorial :)
On my side I still use 0 at reportable change, like that I have only 1 websocket notification, If I use 1 like you I can have 5/6 notifications in 1 second.

closed 2019-04-12 by Smanar