dresden-elektronik/deconz-rest-plugin #1389: Xiaomi sensors become unavailable in Home Assistant after Raspberry Pi device IP is changed due to DHCP settings

jerrychong25, 2019-03-28:
Hi all,

I found out one issue with Xiaomi sensors connection to Home Assistant after Raspberry Pi own device IP is changed due to DHCP settings. 

Originally my Raspberry Pi IP is **192.168.1.160**, after that it changes to **192.168.1.170**.

After IP changed, in Home Assistant these sensors become unreachable, but in deCONZ Phoscon App these sensors are working fine without any issues. 

Home Assistant Error Screenshot:
[image]
[image]

deCONZ Sensor Screenshot:
[image]

To solve this, I forced set Raspberry Pi static IP back to **192.168.1.160**, then everythings working fine again.

Just to confirm, is this error an expected deCONZ behaviour after Raspberry PI device IP is changed?

Thanks.

Kane610, 2019-03-28:
This is not a deCONZ issue. The integration sets the IP on registration, there is no auto updating the IP after that.

I have on my to do to enable updating of IP on gateways further down the line

chimpy, 2019-03-28:
+1 to what @Kane610 said. You should also consider giving out DHCP IP reservations to at least all the smart devices on your network if you are using HA. This will avoid your config breaking because of some arbitrary IP change.

jerrychong25, 2019-03-29:
@Kane610 and @chimpy,

Thanks for the info.

For now, I would set static IP for Raspberry Pi solving this issue.

Thanks.

Kane610, 2019-03-30:
This issue can be closed since the issue has been clarified

closed 2019-03-31 by manup