dresden-elektronik/deconz-rest-plugin #1390: Group setting and scene issue (with Tradfri)

realwax, 2019-03-28:
Hi, 

I created the issue here but as there is not much activity I am not sure if to put it here:
https://github.com/dresden-elektronik/phoscon-app-beta/issues/52

When using the phoscon app beta there are issues I encountered with Tradfri lights and saving individual scenes. The scene itself gets stored but with wrong parameters. It can only be fixed in the old app,
Reproduce:

Create a group, put lights in, set the lights individually, then use the group modification and set them equally, store it as scene. change it to different setting, maybe with individual settings on some lights, store again a scene. Set the scene active and change back to the other. You will notice some lights have not been stored properly. Thsi can be seen in the old app and corrected there.

Thanks

Wolfgang

lbschenkel, 2019-03-31:
Are you sure that you are not being hit by the IKEA firmware bug? One recent example: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1364

realwax, 2019-04-01:
I do not think that this is Ikea firmware ad i can reproduce it with the new web app and solve it with the old. Maybe there is some impact due to fw but the error seem to happen inside deconz as values are stored there...

johnssson, 2019-04-11:
Hi,
I can confirm the issue with Ikea Tradfri (with colortemperature feature). In the changed/new szene, the Brightness is (in my case) correctly stored. But when calling the scene, the color temperature is wrong/doesn't change.
I can also confirm, that this already worked in the past via the old WebApp but since I used the new Phoscon UI, I didn't tried the workaround yet.

I've noticed the bug for over a year and several updates now, but didn't reported it here because I had no github account until now and it worked with my Hue Color bulbs which I have in the room where the scene feature is important for me. Yesterday I updated to the newest version again and the bug persists. So i've searched for this bug here und found this thread. It would be great if someone can fix this,

Thanks, Jonas

closed 2019-08-16 by stale[bot]

lukicsl, 2019-08-22:
I can confirm this issue too