dresden-elektronik/deconz-rest-plugin #1392: ConBee + Ubisys S2-R switches

koffmoff, 2019-03-28:
Hi,

I have a ConBee and a Ubisys S2-R. I cannot get the switches on the S2-R to work in Phoscon. Is the S2-R fully supported or am I doing something wrong?

The two sockets work fine and I can control them as two separate lights. The two switches seem to connect as a single device:

[image]

Clicking on it I get only these options:

[image]

And none of the options above work. If I go to the switch editor, I annot add my switch there:

[image]

Am I doing something wrong?

SwoopX, 2019-03-31:
Hey,

I can confirm this on my S2. Additional observations I could make so far: switching via lights seems to be strange, as the device goes on for about one second and then turns off again. Also, the energy consuption readings I get appear to be incorrect as well. With an 11W light, I have about 600000 of whatsoever being reported after 15 mins.I would have expected to see 3W there. Any thoughts on this?

ebaauw, 2019-03-31:
> I have about 600000 of whatsoever being reported after 15 mins.

You might want to upgrade to the 2019Q1 firmware, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/712#issuecomment-465744443.

> From a device perspective, I see 2 lights, one zhaswitch, one zha power and one zhaconsumption devices available.

That's correct.  The two switch inputs are combine into a single ZHASwitch resource.  Each input controls a different group and `config.group` should contain two groups.  I'm not sure how well Phoscon supports this.

SwoopX, 2019-03-31:
@ebaauw: thanks for mentioning. Checked their page today but only saw 2018/Q1 package. I‘ll try an update and report.

SwoopX, 2019-03-31:
Very well, the gibberish has gone and the values appear to be reasonable now. Thanks again for highlighting!

koffmoff, 2019-04-07:
My S2-R is already at 2019Q1 (0x010E0202). So I guess what I'm seeing is a limitation in Phoscon to handle the switches in S2-R?

ebaauw, 2019-04-07:
I guess so; I'm not too familiar with Phoscon.

koffmoff, 2019-04-10:
Ok, I started playing with the REST API and I can now see my S2-R as a sensor of type ZHASwitch just as you said @ebaauw. It looks like this:

```
"11": {
    "config": {
        "group": "1,2",
        "mode": "momentary",
        "on": true,
        "reachable": true
    },
    "ep": 3,
    "etag": "062185beb595963ac3b8baf4b8a03951",
    "manufacturername": "ubisys",
    "mode": 1,
    "modelid": "S2-R (5602)",
    "name": "S2-R (5602) 11",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "20170503-DE-FB0",
    "type": "ZHASwitch",
    "uniqueid": "00:1f:ee:00:00:00:23:4a-03-0006"
},
```

But how can I access the states of the two switches on the S2-R? There seems to be nothing in the above structure that reflects the switch states?

ebaauw, 2019-04-10:
ZHASwitch resources don’t report a state, only events (press/hold/release).  The last event is reported in `state.buttonevent`.  You need to check `state.lastupdated` to see if the same event occurs twice in a row.

koffmoff, 2019-04-10:
`state.buttonevent` is always `null` and `state.lastupdated` is always `"none"`. I can toggle a connected lamp with the switch so I know it's working. But the `state` field in the sensor remains unchanged no matter what I do. Tried factory reset of the switch and remove/re-add it. Any ideas?

ebaauw, 2019-04-10:
> I can toggle a connected lamp with the switch 

Just to be sure: by “switch” you mean: a traditional wall switch wired to the S2’s input?

I’m not sure how `config.group` got populated when `state.buttonevent` isn’t updated. Best run deCONZ with extensive logging enabled (`--dbg-info=2 --dbg-aps=2 --dbg-error=1`) and check the output for messages containing the S2's mac address 0x001fee000000234a.  I think the binding from the client _OnOff_ and _Level Control_ clusters on the 03 and 04 endpoints to the respective groups might be missing.

koffmoff, 2019-04-10:
Yes, I have a traditional switch connected to (one of) the S2's input. "Cycling" the switch (off/on/off), toggles the state of the output (to which I've connected a lamp). This seems to be the default behavior of the S2.

After resetting the S2 and rejoining it, the `config.group` stays null when toggling the hardware switch. Just as `state.buttonevent`. I haven't configured the S2 in any way after the reset. Just rejoined.

I'm running the docker version which doesn't seem to have an option to set `--dbg-error=1`, but I set `--dbf-info` and `--dbf-aps`. What should I look for in the logs? Maybe this is suspicious ("no button handler"):

> 16:36:06:849 APS-DATA.indication srcAddr: 0x001fee000000234a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -52
> 16:36:06:849   asdu: 18950a00001000
> 16:36:06:850 Node data 0x001fee000000234a profileId: 0x0104, clusterId: 0x0006
> 16:36:06:850 0x001FEE000000234A onOff 1 --> 0
> 16:36:06:850 update ZCL value 0x0006/0x0000 for 0x001FEE000000234A after 4 s
> **16:36:06:850 no button handler for: S2-R (5602) ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000**
> 16:36:06:851 ZCL attribute report 0x001FEE000000234A for cluster 0x0006, ep 0x01
>

ebaauw, 2019-04-10:
> 16:36:06:850 no button handler for: S2-R (5602) ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000

That's the attribute report for the switch's first output (endpoint 0x01, the `/lights` resource's `state.on`).  You want a message like that for ep 0x03 (first input) or 0x04 (second input) and cmd 0x02 (Toggle).  If you don't see any of those, you might need to bind the outputs to their groups manually (using the _Bind Dropbox_ panel in the deCONZ GUI).

>  "Cycling" the switch (off/on/off), toggles the state of the output (to which I've connected a lamp).

Sounds like you connected a rocker switch, that has an on vs off state, instead of a momentary (or pulse) switch, that only delivers power while it's pressed.  You'll want to set `config.mode` to `"rocker"`, so you don't have to do the off/on/off routine.

koffmoff, 2019-04-10:
Yes it's a rocker switch. Setting `config.mode` to `"rocker"` sure made its behavior more natural! Thanks! I also used Phoscon to assign groups to the switch. Is that what you meant with "using the Bind Dropbox panel in the deCONZ GUI"?

Now the switch looks like this:

```
{
    "config": {
        "group": "1,1",
        "mode": "rocker",
        "on": true,
        "reachable": true
    },
    "ep": 3,
    "etag": "9eb299eb7b9e4e14ffbc35a49155f7c6",
    "manufacturername": "ubisys",
    "mode": 1,
    "modelid": "S2-R (5602)",
    "name": "S2-R (5602) 10",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "20170503-DE-FB0",
    "type": "ZHASwitch",
    "uniqueid": "00:1f:ee:00:00:00:23:4a-03-0006"
}

```
`config.group` is now assigned. But `state.buttonevent` stays `null` and no sign of any `ep: 0x03` or `ep: 0x04` in the logs.

ebaauw, 2019-04-10:
> Is that what you meant with "using the Bind Dropbox panel in the deCONZ GUI"?

No.  In the deCONZ GUI (the graphical user interface under X11 or Windows that runs on the same machine as deCONZ), there's a panel where you can drag&drop a source cluster and specify a destination group to create a binding.  See e.g. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/978#issuecomment-474050373.

For that, you need to start deCONZ with the GUI enabled - I don't how how that's done in a Docker setup, or if that's even possible.

closed 2019-08-15 by stale[bot]