dresden-elektronik/deconz-rest-plugin #1396: Aqara Vibration Sensor Set Sensitivity

m50434, 2019-03-30:
Hi,
I just add the Aqara Vibration Sensor (https://www.gearbest.com/goods/pp_009661787808.html?lang=en&wid=1433363) and wonder how I can set the sensitivtiy-value in the phoscon-App.  In the original Xiaomi-App there are there values for sensitivity: low, medium and high.
At the moment it is obviously not supported in phoscon-app. Is it planed for future?!

chimpy, 2019-04-01:
It is possible to set this via the API. I can't remember how though. Search the issues here.

gpl1998, 2019-04-01:
curl -H 'Content-Type: application/json' -X PUT -d '{"sensitivity": 1}' http://yourIP/api/yourKEY/sensors/yourSensorID/config

Sensitivity is from 1 to 21 where 1 is the most sensitive.

m50434, 2019-04-01:
Great. That worked.
Thanks.

closed 2019-04-01 by m50434

ladefoged81, 2019-04-01:
I just did the same for a Hue motion sensor.
The REST API guide is here: http://dresden-elektronik.github.io/deconz-rest-doc/

gekomax, 2019-06-17:
Hi, where I can find the api key? is the number wrote in the HA integration tab?

Stig1987, 2019-07-04:
I got it working with help of gpl1998 comment above despite no pre-knowledge. Below some hints that had saved time for me:

1) Finding API key and understanding the concept: I used the guide [http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/], I used the firefox rest client [https://addons.mozilla.org/de/firefox/addon/restclient/]

2) Understanding how to fetch and set data: [http://dresden-elektronik.github.io/deconz-rest-doc/sensors/]

3) Getting yourSensorID right: I tried with the name I choose for the sensor inte the Phoscon APP. but in my case it was a number, "60". Not all numbers where linked to sensors so try and error was not successful for me. Instead, I sent a GET to http://<ip>:<port>/api/<api-key>/sensors , downloaded the big chunc of text, opened it in visual studio code, searched for the sensor name, in my case "back_door_vibration" and sorted out which chunc of text that belonged to this sensor. The number in the beginning, was then the id.

The chunc of thxt for my vibration sensor:
`"60":{"config":{"battery":98,"on":true,"pending":[],"reachable":true,"sensitivity":11,"sensitivitymax":21,"temperature":2500},"ep":1,"etag":"31b154ca59c1ea415ddeba9177089e01","manufacturername":"LUMI","mode":1,"modelid":"lumi.vibration.aq1","name":"back_door_vibration","state":{"buttonevent":1009,"lastupdated":"2019-07-04T19:39:22"},"swversion":"20180130","type":"ZHASwitch","uniqueid":"00:15:8d:00:02:c4:e5:0c-01-0101"},"

4) Understanding what command to send: I used "Method" PUT, URL "http://<ip>:<port>/api/<api-key>/sensors/<sensor id> and "Body" { "sensitivity": 1 } in firefox rest client. I think the command in gpl1998 includes all of this for some more command-line like format

No guarante that I have got it right but the response on GET is now at least ""sensitivity":1". I hope this comment (my first on GitHub) is recieved well and can save time for someone.

HansOHG, 2019-07-26:
Hello Stig and thank you for the explanation.
I have managed to get the api and the sensor id for my vibration sensor but when I try to send the PUT command with body:  { "sensitivity": 1 }
I get a error message: 
`[
    {
        "error": {
            "address": "/sensors/sensitivity",
            "description": "parameter, sensitivity, not available",
            "type": 6
        }
    }
]`

Any thoughts what I might be doing wrong?

Smanar, 2019-07-27:
Are you sure for the id ?

HansOHG, 2019-07-27:
My sensor has the id: 20 if I understand correct.

[image]
[image]

ebaauw, 2019-07-27:
I wouldn’t try and make sense of the error message, but it does indicate an error: you need to PUT to `/api/xxxxxx/sensors/20/config`

Smanar, 2019-07-27:
Yep, you are using bad url, your is finishing by "sensors/20"

HansOHG, 2019-07-27:
Thank you guys. 
adding /config to the end of the url did work. My sensitivity is now 1

00lex, 2019-08-25:
I spent 2h on testing and this doesn't work! 
I used a servo on different speeds for movement and glued the sensor on a speaker membrane for testing impulse with a 50hz tone on various volumes.

the results are identical. sensitivity 1 = 11 = 21

Smanar, 2019-08-25:
Put the sensor on a table, and knock the table with different sensitivity.

00lex, 2019-08-25:
Knocking on a table is not precise enough and not reproducible. 
because of that I use the Membran of a speaker to get exact the same intensity. 
If play/pause a 50hz tone, you get every time the same knock which you can de increase by volume

MHOMEASS, 2020-08-28:
> Knocking on a table is not precise enough and not reproducible.
> because of that I use the Membran of a speaker to get exact the same intensity.
> If play/pause a 50hz tone, you get every time the same knock which you can de increase by volume

I had the same experience. I set the sensitivity to '1' and '21' but didn't notice any difference. When I looked at the status of the senor I saw that 'sensitivity' was 'pending'. Then I read a post that you have to press the reset button for a few times for 1 second directly after you set the sensitivity. The pending status disappeared and know the sensitivity appeared to be more sensitive.

Spanishu, 2020-11-10:
I have just updated my Sensor according to the instructions. However, I have had the same experience than @MHOMEASS and there is not difference in the real sensitivity of the sensor. The sensitivity is 21. Any suggestions?

Smanar, 2020-11-10:
Are you testing the vibration or the tilt ?

Spanishu, 2020-11-11:
I have changed the value to 1 and it is much more sensitive. Not as much as I would like but it works!

fr1cklr, 2023-04-23:
Please help me!

> I had the same experience. I set the sensitivity to '1' and '21' but didn't notice any difference. When I looked at the status of the sensor I saw that 'sensitivity' was 'pending'. Then I read a post that you have to press the reset button for **a few times for 1 second** directly after you set the sensitivity. The pending status disappeared and now the sensitivity appeared to be more sensitive.

What exactly means "**a few times for 1 second**"?

I can't get rid of the ""pending":["sensitivity"]" regardless how often I press the button :(

Smanar, 2023-04-24:
If i m right, you can just set the value you want, and wait for deconz be able to change it, can take 1 hour.

Mimiix, 2023-12-07:
These questions don't belong here. Please use the forums.