dresden-elektronik/deconz-rest-plugin #1397: phozcon gateway: firmware not connected and unable to change channel

02dag, 2019-03-31:
I have installed the raspbee on hassio x64 on rpi 3 B+

The device is recognized in the log. When connecting to the web gui I can login to the phozcon gateway and I see that it say firmware not connected. It is als not possibe to change channell to 11 as an example. When refreshing the page it resets to channel 0.

Firmware is at 2.05.60 
Can connect to it with vnc. In VNC it also says Not connected

I am running the addon from marthoc but the same happened with the addon from dresden.

These errors have been reported by many people, but no answer has been given to the problem yet. Plenty of workarounds though, but none that I can use.

Must be bad code somewhere or a permission issue,my guess anyway.

manup, 2019-03-31:
Looks like serial communication with RaspBee is not up.
Please check that the serial port is configured properly as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

Philje123, 2019-04-01:
Have you updated the firmware on the conbee? I first did this via a windows PC using GCFlasher.

In Hass.io go to system and click "Hardware" and make sure you have put the correct path to the conbee in the add-on settings.

02dag, 2019-04-01:
This is a raspbee, and I have inserted /dev/ttyAMA0 as stated in my hardware settings overview.

This is also hassio image. Could it be that it is not supported on this platform?

If not supported but supposed to work, how do I check the serial config here?

Load them as options in the config-file? Someone knows what to insert there?

Philje123, 2019-04-01:
Did you also add this to your raspberry pi config.txt?

enable_uart=1
dtoverlay=pi3-disable-bt

Philje123, 2019-04-02:
No it's the other way round. That's required for the raspbee.

The conbee doesn't required any additional changes since it works via USB.

It's because the conbee is via the GPIO pins and there's a conflict I believe between the onboard Bluetooth. So I think you can't use Bluetooth with rapsbee which is why I went for the conbee.

02dag, 2019-04-04:
Finally got to changing this, and now runs with bluetooth disabled. Got a different log now then before:

rmmod: ERROR: Module usbserial is builtin.
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
RaspBee: /dev/ttyAMA0 (default)
no FTDI devices found
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0libpng warning: iCCP: known incorrect sRGB profile

100  2734  100  2734    0     0   7655      0 --:--:-- --:--:-- --:--:--  7679
1135-0000-201000A0-FLS-PP3_RGBW.zigbee: OK
1135-0000-2010009B-FLS-PP3_RGBW.zigbee: OK
1135-0006-20100021-Ribag.zigbee: OK
14010400_0x0005.zigbee: OK
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()

Not sure what all this means. It says zigbee OK, but also an error on usbserial. In the wbeUI it still says firmware not connected and when trying to change the channel this will not work.

02dag, 2019-04-08:
Finally I gave up and installed the rpi-image which is downloadable directly from dresden elektroniks with Raspbian and support for raspbee. This worked right away. I can see there that the raspbee got the address /dev/ttyS0. My meter which is a zigbee module from Develco connected to the raspbee.

xbmcnut, 2019-06-13:
> 
> 
> Have you updated the firmware on the conbee? I first did this via a windows PC using GCFlasher.
> 
> In Hass.io go to system and click "Hardware" and make sure you have put the correct path to the conbee in the add-on settings.
> [image]

Can you please detail how you got Windows to flash the Conbee (II)? Windows deConz see's my stick but when I run version 2.10 with `GCFFlasher -l`, I get "no FTDI devices found" so I'm unable to upgrade.

manup, 2019-06-17:
For Windows normally the firmware update button should be shown under 
*Phoscon App > Menu > Settings > Gateway*

Boh1, 2019-09-01:
Same issue here. Can't change Zigbee channel and there is no firmware update button. Windows 10, deCONZ_Setup_Win32_V2_05_67.

xbmcnut, 2019-09-01:
I had the same issue after updating to deCONZ 3.3. I had my device defined as `/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_xxxxxxx `which was not being recognized so I changed the path to `/dev/ttyACM0` and it started communication again and the Zigbee channel came back. I then reset the path back to `/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_xxxxxx `and it's all working again. Possible that I also threw in a deCONZ or Hassio restart in there too.

* Hassio on NUC with Conbee II | Conbee II is running  26490700 *

closed 2020-01-06 by stale[bot]

marine1988, 2020-03-15:
im facing ths issue one hassio with docker!